PLC-lang/rusty #114: What's the goal of this project ?

mengzhisy, 2021-01-04:
Hi, this project looks very interesting, but I am not sure what the ultimate goal of this project is. Will this be a long-term maintenance project? Or is it just for learning, like most rust projects?

ghaith, 2021-01-04:
Hello,
[Structured Text](https://en.wikipedia.org/wiki/Structured_text) is a language used in Industrial automation.
It is part of the [IEC 61131-3](https://en.wikipedia.org/wiki/IEC_61131-3) Standard
This is our attempt at making a compiler for that language based on LLVM. 
We are using it mostly as a learning project for Rust/LLVM and compiler design in general but we would like it to eventually become production ready. 
Regarding maintenance, we are currently doing this in our free time so no promises. We released it under LGPL to encourage people to contribute to it.
We obviously hope a community can grow around it, but it is a niche product with a small market.

riederm, 2021-01-04:
It started as a learning project, but from my perspective it eventually grew out of it.
We're adding features that go beyond the pure learning purpose already - e.g. trying to fully comply to the IEC61131-3 standard.

As ghaith already explained, this is a free-time project - but is LGPL so anybody can maintain it. To enable comunity contributions we're still lacking a ton of documentation.

As for now we gained some new velocity over the holidays so we plan to carry this into the next year and soon'ish have a feature-complete ST-compiler based on LLVM.

riederm, 2021-01-04:
@mengzhisy what exactly is your interest? are you looking for an ST-compiler? or more from a technological point of view?

mengzhisy, 2021-01-04:
Thank you for your explanation. I am more interested about rust/llvm itself. Recently, I was investigating whether llvm can accelerate data analysis and processing. However I am a beginner to llvm, so I hope the rust/llvm community will become more and more active. Finally, I hope this project will get better. (Respect for your efforts !)

mengzhisy, 2021-01-04:
I will close this issue ^_^

closed 2021-01-04 by mengzhisy