PLC-lang/rusty #115: Avoid mod.rs as a filename in modules to increase maintainability and readability

riederm, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
I find it hard to work with so many mod.rs files inside the same project. I'd like to rename the mod.rs in the different modules, to the module's name so it is easier to locate a certain line or method.

**Describe the solution you'd like**
Rename mod.rs to [module-name].rs

**Describe alternatives you've considered**
cut the project into seperate libraries - this makes development / local dev-setup a bit harder

ghaith, 2021-01-04:
https://www.reddit.com/r/rust/comments/b65ecn/modrs_in_rust_2018_yes_or_no/
Looks like we can have codegen.rs and then a codegen/ dir with submodules.

ghaith, 2021-01-04:
See also : https://doc.rust-lang.org/reference/items/modules.html
```
Note: Previous to rustc 1.30, using mod.rs files was the way to load a module with nested children. It is encouraged to use the new naming convention as it is more consistent, and avoids having many files named mod.rs within a project.
```

closed 2021-01-19 by ghaith