PLC-lang/rusty #125: Initial values for functions

ghaith, 2021-01-19:
**Is your feature request related to a problem? Please describe.**
#123 added initial values support, but we skipped it for functions.
This issue is to keep track of that.

**Describe the solution you'd like**
If a function declares inital values for VAR values, they should be taken into account.
We also need to define correct behaviour on INPUT variables (Not sure if they are optional for functions)

riederm, 2021-01-23:
is there a disadvantage if we initialize the VAR-variables at the beginning of the function body? I feel this would be the easiest solution.

ghaith, 2021-01-23:
I think that makes sense we already initialise some local variables at this
point

riederm, 2021-01-23:
complex initialization of arrays and structs will make some problems maybe. Are these allowed as normal statements? we'll see.
scalars should work for now

ghaith, 2021-01-23:
> complex initialization of arrays and structs will make some problems maybe. Are these allowed as normal statements? we'll see.
> scalars should work for now

My gut feeling about this is to go with function calls for complex initializer, so we generate an initializing function and call it where we need to. But we should probably open a new issue to track that

PR #126 (2021-01-25): Initial values for functions
generate an assignment statement inside the function for every VAR
with an initializer

Fixes #125
commit 78c4dbe8 (2021-01-25): Initial values for functions (#126)

generate an assignment statement inside the function for every VAR
with an initializer

Fixes #125
closed 2021-01-24 by riederm