PLC-lang/rusty #128: Support for Date/Time types

ghaith, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
The standard supports they types DATE TIME and DATETIME
We should be able to define parse and generate these types

**Describe the solution you'd like**
Parsing and initialization should follow the standard
For codegen i would think we should just store a Unix time value in a long value.

**Describe alternatives you've considered**
N/A

**Additional context**
This is a placeholder issue for all types Sub issues should be created for individual types and support
[] DATE
[] DATETIME
[] TIME

riederm, 2021-04-22:
@ghaith Date- & Time-Literals would be optoins to use logos-callbacks to transform then into numbers during lexing.

do you have any feelings regarding callbacks? I'm unsure about it, because ...
- so far, the lexer does only create tokens, no behavioral code whatsoever
- it looks like I could do better error-reporting inside the parser (e.g. if you write the 32nd of April)

ghaith, 2021-04-23:
The advantage of the lexer is it might be easier. But otherwise the parser wins here. It's also logically the place to put it I think.
Regarding error reporting while lexing, we could eventually think of adding more Error tokens with different properties but for now it feels better to stick with the parser and not have the lexer also parse things.

closed 2021-05-19 by ghaith