PLC-lang/rusty #138: Action Support

ghaith, 2021-03-09:
**Is your feature request related to a problem? Please describe.**
A PLC Program's (private) interface should be reusable in other functions by defining an Action

**Describe the solution you'd like**
### Defining Actions
An Action can be defined and bound to a POU, the action would have a body and can be called to execute its code.
The action is bound to the POU in 3 different methods : 
- An ACTIONS container is defined directly after a POU `PROGRAM MainProg END_PROGRAM ACTIONS` will bind all its actions to the POU
- An ACTIONS container with the POU name `ACTIONS MainProg` will bind all actions  that POU
- An ACTION with the POU name as a qualifier `MainProg.MyAction` will be bound to that POU

### Calling Actions
An action can be called inside its parent POU by specifing its name as a Function folowed by open and close brackets. `myAction`
An action can be called from anywhere by its fully qualified name followed by open and close brackets `MainProg.myAction()`
Parameters of an Action are optional, an Action will use it's containg POU or instance as parameters if none are provided.

**Additional context**
VAR_IN_OUT and variable access on actions instead of containers are not considered here.

ghaith, 2021-03-14:
@riederm I'm considering dropping the first case : 
"An ACTIONS container is defined directly after a POU `PROGRAM MainProg END_PROGRAM ACTIONS` will bind all its actions to the POU" because it will add much more complexity to the parser in scenrarios where the POU directly before you is another related actions block:
```
PROGRAM foo
END_PROGRAM
ACTION foo.bar
END_ACTION
ACTIONS
  ACTION buz <<<<<
  END_ACTION
END_ACTIONS
...
```
Or should we just solve this as best effort and if whatever before you was an action we go into undefined behaviour/crash?

riederm, 2021-03-22:
> Or should we just solve this as best effort and if whatever before you was an action we go into undefined behaviour/crash?

I agree, we should drop this. It makes not much sense in a setup where we dont enforce one POU per file

PR #157 (2021-04-01): Actions Support
Added support for actions

Fixes #138
closed 2021-04-01 by riederm