PLC-lang/rusty #184: Add Linker support

ghaith, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
We currently rely on Clang to link our object files, We should use the LLVM LD API instead.

**Describe the solution you'd like**
By default, always attempt to link the output as an executable.
Hide the normal compile/object code generation behind a -c flag
When a -shared option is passed, link the file as a shared library.
Respect targets when linking
Add a --sysroot option to provide the correct sysroot when compiling / linking for other targets.

**Describe alternatives you've considered**
If we skip this we would need to always link using a call to ld or clang, but this is also an option.

ulmer-a, 2021-07-20:
Now that it seems like it's not possible to call the llvm linker directly from rusty without writing object files to disk -- should we implement the call to the system's lld or clang?

ghaith, 2021-07-20:
I think we should still go with the same plan, and save intermediate files in a temp or work directory.
A call to the system's lld/clang means a dependency on some installed clang/lld version on the user's system, which means we have to be careful if the llvm version they are using is different from the one we support. Going with the previous plan of programmatically calling the linker would mean we have all llvm dependencies contained in our release

closed 2021-08-29 by ghaith

ghaith, 2021-08-30:
We didn't add a sysroot option , so cross compile won't work yet. I'll open a new issue for that.

PR #243 (2021-08-29): Support linking on Linux
This PR will close #184 and provide a linker driver built-in. The documentation is updated as well. New command line options:
* `-c` Skip linking. You'll be left with an object file.

- [x] Implement linker driver for Linux
- [x] Add `-c` cmdline option
- [x] Update default filenames for executables
- [x] Update documentation
- [x] Do some manual testing on Linux
- [ ] Add some more tests
- [x] Command line option for "add library"
- [x] Command line option for "add include path