PLC-lang/rusty #208: Support for CONSTANTS

ghaith, 2021-07-23:
**Is your feature request related to a problem? Please describe.**
Some variables can be marked as constants, and would then be unchangeable in the code.
The Standard marks such blocks with `VAR CONSTANT`
Global and other blocks can also be marked constant.

**Describe the solution you'd like**
Constants are generally stored in the AST as normal variables.
Writing to a constant would generate a diagnostic (Optional for now, but we should keep that in mind).
We could track assignment rules, but not indirect assignments through pointers.

We should try to replace the constants with actual values after the AST has been generated, possibly during indexing.

Constants can be used instead of sizes or ranges in data types or in case statements/ for statements.
i.e. Wherever a literal is expected.
e.g.
```
VAR CONSTANT
  consta : INT := 5;
  constb : INT := consta;
END_VAR

VAR
  x : INT := consta; // Should evaluate to x : INT := 5;
  y : STRING[constb]; //Should evaluate to y : STRING[5];
END_VAR
```

**If we decide to not replace constants, we need to extend the index to support getting values of the constants so we can use them during datatype generation.**

ghaith, 2021-08-31:
We discussed a proposal for solving this during the pre-processing phase.
Should we go with that strategy? This involves navigating the AST at least twice unless we make certain elements of the AST modifiable (Initializers).
Another strategy was to hold the resolved constant values in the index after we collect them all. This way replacing the values could be solved as lookups on the values map.

If we wrap the initializers (and references?) of the AST in [Cells](https://doc.rust-lang.org/std/cell/), we could probably save these in the map instead and modify them during preprocess.

riederm, 2021-09-04:
I dropped my initial idea of solving the problem at codegen time by just switching to the constant's initializer-AST for generation. We need to precisely know the sizes of types even when constants are used in the declaration: 
```
VAR_GLOBAL CONSTANT
   LEN : INT := 10;
END_VAR

VAR_GLOBAL
  myArr: ARRAY[0..LEN] OF INT;
END_VAR
```

So i gonna evaluate the constant's at compile time now. The Question is how far we need to persue this ... 
e.g.
```
VAR_GLOBAL CONSTANT
   LEN : ARRAY[0..2] OF INT := [10, 20, 30];
END_VAR

VAR_GLOBAL
  myArr1: ARRAY[0..LEN[0]] OF INT;
  myArr2: ARRAY[0..LEN[1]] OF INT;
  myArr3: ARRAY[0..LEN[2]] OF INT;
END_VAR
```
And we could even think about initialized constant structs?