PLC-lang/rusty #216: Allow _ as digit-separator in numeric literals

riederm, 2021-07-29:
allow "_" to make numbers more readable for any number-format:

1_000_000
0#DEAD_BEAF
2#0000_0011
8#77_00

The _ should be allowed at any position in the number-literal and not change the number at all.
Two "_" in a row should not be allowed: 1_000__000 should fail

ulmer-a, 2021-07-29:
Oops... I think I forgot to include this when dealing with #205... This shouldn't take long.

ulmer-a, 2021-07-29:
Any idea on why our lexer library will match `16#([0-9a-fA-F]_)*[0-9a-fA-F]+` but not `16#([0-9a-fA-F]+_)*[0-9a-fA-F]+` ?  The latter one would solve my problem not matching hex literals with double underscores...

riederm, 2021-07-29:
if there is no easy solution to this problem, I suggest we put it back and wait and see if the logos-guys maybe find a solution in the next months...

ulmer-a, 2021-07-29:
Ok. A temporary workaround could be to match for any '_' and deal with duplicate underscores within the parser.  But I agree to put it back since the workaround wouldn't be very elegant.

ulmer-a, 2021-07-30:
I'm probably going to implement it anyway, but for the moment allow multiple consecutive underscores. That way we're not going to run into compatibility issues and when the logos guys have fixed their stuff, we can just go ahead, change the regex so that it only matches single underscores and that's it!

PR #217 (2021-07-30): Underscores in Integer literals
This will close #216 

This PR will allow literals to contain underscores as described in the linked issue. However, since the regex implementation of logos has a bug, this PR will for now also allow for any number of underscores at any location in the literal. On one side, this will allow code with underscores in literals to compile, but on the other hand will lead to a panic in the case that the literal consists of *only* one or more underscores. Fortunately, to fix this completely, we only have to swap out 4 regular expressions once Issue [logos#213](https:// [... cut]
closed 2021-07-30 by ulmer-a