PLC-lang/rusty #219: Support Classes - part1: parse classes

riederm, 2021-07-30:
Parse class POUs and get an approriate representation in the AST.

Classes ... 
- have an identifier
- declare multiple variable declaration blocks
- dont have a body
- can have mulitple methods

Methods... 
- have an identifier
- can have a visibility modifer (PUBLIC, PRIVATE, INTERNAL, PROTECTED [default])
- may declare additional variable delcaration blocks
- can declare a return-type (like functions do)
- have a body

```
CLASS COUNTER
	VAR
		CV: UINT;
		Max: UINT := 1000;
	END_VAR
	
	METHOD PUBLIC Up: UINT
		VAR_INPUT INC: UINT; END_VAR
		VAR_OUTPUT QU: BOOL; END_VAR
		
		IF CV <= Max - INC THEN
			CF := CF + INC;
			QU := FALSE;
		ELSE
			QU := TRUE;
		END_IF
		UP := CV;
	END_METHOD
	
	METHOD PUBLIC Up5: UINT
		VAR_OUTPUT QU: BOOL; END_VAR
		Up5 := THIS.UP(INC := 5, QU => QU);
	END_METHOD
END_CLASS
```

ulmer-a, 2021-08-03:
The grammar I pulled out of the standard looks something like this:

```
CLASS [ABSTRACT|FINAL] <ClassName>
	[USING <Namespace1>, <Namespace2>, ... ]
	[EXTENDS [...[<Type2>.]]ClassName]
	[IMPLEMENTS <Interface1>, <Interface2>, ...]
	
	
	// Any number of VAR / VAR (NON_)RETAIN blocks
	VAR [CONSTANT|RETAIN|NON_RETAIN] [PUBLIC|PROTECTED|PRIVATE|INTERNAL]
		<VarName> : ... ;
	END_VAR
	
	// Any number of VAR_EXTERNAL blocks
	VAR_EXTERNAL [CONSTANT] 
		<VarName> : ... ;
		...
	END_VAR
	
	// Any number of "LocPartlyVarDecls"
	VAR [RETAIN|NON_RETAIN]
		<VarName> AT % [I|Q|M] * : ... ;
	END_VAR
	
	// Any number of method declarations like this:
	METHOD [PUBLIC|PROTECTED|PRIVATE|INTERNAL] [FINAL|ABSTRACT] [OVERRIDE] <MethodName> [: RETURN_TYPE]
		
		// Any number of VAR blocks
		VAR [CONSTANT] [PUBLIC|PROTECTED|PRIVATE|INTERNAL]
			<VarName> : ... ;
		END_VAR
	
		// Any number of VAR_EXTERNAL blocks
		VAR_EXTERNAL [CONSTANT] 
			<VarName> : ... ;
			...
		END_VAR
		
		/// Any number of IO_VAR declarations^
		[VAR_INPUT|VAR_OUTPUT|VAR_INOUT] [RETAIN|NON_RETAIN]
			...
		END_VAR
		
		VAR_TEMP
			...
		END_VAR
		
		<function_body>
	
	END_METHOD

END_CLASS
```

ulmer-a, 2021-08-05:
**Variable Blocks**

Within a class declaration:
* `VAR_EXTERNAL` and `VAR_EXTERNAL_CONSTANT` are used define global variables only accessible from within the class.
* `VAR` blocks are member variables and can be marked public, private, protected or internal.
* blocks marked `RETAIN` or `NON_RETAIN` are member variables with special behaviour with respect to a reset.

Within a method declaration:
* ...

ulmer-a, 2021-08-06:
**Regarding Variable Blocks inside a method declaration:**

* `VAR_INPUT`, `VAR_OUTPUT` and `VAR_INOUT` can be used to specify arguments to the method. Only the keywords `RETAIN` or `NON_RETAIN` may be specified on those.
* `VAR` and `VAR_TEMP` blocks can be used to declare temporary variables. Their use is equivalent.

ulmer-a, 2021-08-06:
The parser will parse all variable blocks equally, so the validity of specified keywords and initializers, etc. has to be checked during semantic analysis. However, specifying a IO blocks inside a class declaration will yield a parser error.

PR #225 (2021-08-12): Parse 61131-3 Classes
After this PR, rusty will be able to parse classes. Closes #219. Please merge PR #211 first.

Note that I accidently rebased the changes from PR #211 into this branch (first 10 commits below). Would have been better to merge it... It's going to be squashed anyway, though.

Deviation from the standard: We do not support `VAR_EXTERNAL` constructs since they are rarely used. Namespaces, Inheriting and interface have not been implemented, yet.
closed 2021-08-12 by ulmer-a