PLC-lang/rusty #245: VAR_TEMP support

ghaith, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
VAR_TEMP are variable blocks that are re-initialized on every call.
They are similar to a function variable.

**Describe the solution you'd like**
Allow parsing VAR_TEMP.
When generating code, assign all the values of a temp block to their initial values.

riederm, 2021-09-24:
to what extent does `VAR_TEMP` and `VAR` differ?

riederm, 2021-09-24:
todo: add VAR_TEMP testcase to 
`src/validation/tests/variable_validation_tests.rs: constant_on_illegal_var_blocks_cause_validation_issue`

ghaith, 2021-09-25:
> to what extent does `VAR_TEMP` and `VAR` differ?

In FUNCTIONS they are the same, otherwise TEMP blocks behave like a function's VAR block, they get re-initialized on every call to that POU