PLC-lang/rusty #247: Literal type cast.

ghaith, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
A literal can be written in the form of `DATATYPE#XXXX`
This makes a literal of the given type.
Example
```st
x := INT#1234;
```
This will force 1234 to be treated as INT instead of DINT

**Describe the solution you'd like**
We should parse this correctly and enrich the AST with that information on literals

**Additional context**
`CHAR` might also need to be covered, we need to review the standard for that.

riederm, 2021-08-18:
the norm mentions following examples:
- INT#-123
- INT#16#7FFF
- WORD#16#AFF
- WORD#1234
- UINT#16#89AF
- CHAR#16#41
- BOOL#0
- BOOL#1
- BOOL#FALSE
- BOOL#TRUE
- STRING#'OK'
- CHAR#'X'
- WSTRING#"OK"
- WCHAR#"X"
- WSTRING#'OK'
- WCHAR#'X'
- Traffic_light#Red (Enum Literal)

I think we should support all literals (bools, chars, all ints, all reals, all strings, enums)
Date & time literals already require such a prefix, so nothing to do here

riederm, 2021-08-25:
I have a conceptual problem.

the ast cannot store a literal > i64
so with this AST you cannot have the full range of u64 (ULINT) literals in your sourcecode

e.g.: `ULINT#16#FFFF_FFFF_FFFF_FFFF`

how much of an overkill is it, if we store the LiteralInteger either as an i128?

riederm, 2021-08-25:
should we allow this?
'INT# 123;'

riederm, 2021-08-25:
> I have a conceptual problem.
> 
> the ast cannot store a literal > i64
> so with this AST you cannot have the full range of u64 (ULINT) literals in your sourcecode
> 
> e.g.:
> 
> * `ULINT#16#FFFF_FFFF_FFFF_FFFF`
> * `ULINT#-1`
> 
> how much of an overkill is it, if we store the LiteralInteger either as an i128?
> what do you think @ghaith?

in a call we agreed on treating literals as i128

commit 65c44fda (2021-08-22): support literals with type-prefixes

supports numeric prefixes and boolean prefixes
support prefixed, negative numerics
support true, false, 0 and 1 for boolean

closes #247
commit 9c91b8c4 (2021-08-24): support literals with type-prefixes

supports numeric prefixes and boolean prefixes
support prefixed, negative numerics
support true, false, 0 and 1 for boolean

closes #247
commit 61444e23 (2021-08-29): support literals with type-prefixes

supports numeric prefixes and boolean prefixes
support prefixed, negative numerics
support true, false, 0 and 1 for boolean

closes #247
PR #260 (2021-08-31): Typed literals (INT#123, Color#RED, BOOL#1, ...)
closes #247
closed 2021-08-31 by ghaith