PLC-lang/rusty #256: make case insensitive parser configurable

riederm, 2021-08-22:
The parser is currently case-insensitive due to #255.
The parser behavior should be configurable from the cli (like strict mode).

ghaith, 2022-06-22:
The lexer is configured at compile time, we can maybe make that a feature
We can make the reference check configurable (we currently store everything in lower case, and search for lowercase matches in the index)

teksturi, 2024-11-26:
It is also possibility that there is two different lexer. That can be then configured runtime which one to use.

Case insensitive lexer can be then mapped to case sensitive lexer. That way case sensitive lexer would be really fast but case insensitive lexer would be even more slower.

Also other way around is possible and in that case probably both are about same speed as current case insensitive lexer.