PLC-lang/rusty #257: performance-analysis: case-insensitive parser

riederm, 2021-08-22:
measure the performance-penalty of the case-insensitive parser introduced in #255 using huge files.

volsa, 2023-07-11:
I did some quick and dirty benchmarking to see if there are any performance hits using a case-insensitive lexer and as far as I can tell there aren't any (at least none you'd feel). The corpus for the benchmark consists of all ST files in our repository, saved in one file via `find . -type f -name '*.st' -exec cat {} + > out.txt`. That adds up to almost ~15k lines, which when running the lexer in release mode like so
```Rust
let now = Instant::now();
while !lexer.is_end_of_stream() {
    lexer.advance();
}
let elapsed = now.elapsed().as_micros();
```
returns the following

- Insensitive
  ```
  Took 722us (End; SourceRange { range: 276576..276588 }; 0)
  Took 681us (End; SourceRange { range: 276576..276588 }; 0)
  Took 660us (End; SourceRange { range: 276576..276588 }; 0)
  Took 629us (End; SourceRange { range: 276576..276588 }; 0)
  Took 762us (End; SourceRange { range: 276576..276588 }; 0)
  Took 656us (End; SourceRange { range: 276576..276588 }; 0)
  Took 650us (End; SourceRange { range: 276576..276588 }; 0)
  Took 646us (End; SourceRange { range: 276576..276588 }; 0)
  Took 632us (End; SourceRange { range: 276576..276588 }; 0)
  Took 644us (End; SourceRange { range: 276576..276588 }; 0)
  Average: 668us
  ```
- Sensitive
  ```
  Took 633us (End; SourceRange { range: 276576..276588 }; 0)
  Took 593us (End; SourceRange { range: 276576..276588 }; 0)
  Took 566us (End; SourceRange { range: 276576..276588 }; 0)
  Took 558us (End; SourceRange { range: 276576..276588 }; 0)
  Took 732us (End; SourceRange { range: 276576..276588 }; 0)
  Took 577us (End; SourceRange { range: 276576..276588 }; 0)
  Took 541us (End; SourceRange { range: 276576..276588 }; 0)
  Took 573us (End; SourceRange { range: 276576..276588 }; 0)
  Took 602us (End; SourceRange { range: 276576..276588 }; 0)
  Took 624us (End; SourceRange { range: 276576..276588 }; 0)
  Average: 599us
  ```
Maybe I'm missing something as these numbers are kinda surprising to me to be honest; anyways here's the code https://github.com/PLC-lang/rusty/compare/master...__bench_lexer

Follow-up: Is there any interest in creating a metric for the lexer in case we implement a hand-written lexer at some point in the future for whatever reason? Also this section seems obsolete https://plc-lang.github.io/rusty/arch/parser.html#discussion-rusty-lexer? 

cc @ghaith @riederm

riederm, 2023-07-12:
I agree that the time spent in the lexer seems insignificant, so lets not spend too much effort here. 
Lets close the discussion - https://plc-lang.github.io/rusty/arch/parser.html#discussion-rusty-lexer - i'll do that.

I initially wanted to shrink the lexer because the logos token-macro became so big, rust-analyzer had some issues when you opened the file in vs-code. It looks like this issue disappeard.

volsa, 2023-07-12:
Great, closing this issue then. 
Removing the discussion is tracked in PR https://github.com/PLC-lang/rusty/pull/902

closed 2023-07-12 by volsa