PLC-lang/rusty #258: Use derivable/reproducable names for derived datatypes

ghaith, 2021-08-23:
**Is your feature request related to a problem? Please describe.**
We currently generate datatype names for custom types based on the variable name and then name of its container.
While this provides unique names, they don't help with finding compatible types
For example, given the following declaration: 
```st
PROGRAM prg
VAR
  x : ARRAY[0..1] OF INT;
  y : ARRAY[0..1] OF INT;
END_VAR
...
```
The types x and y are compatible, but since we will create two different types (`__prg_x` and `__prg_y`) we will treat them as separate.

**Describe the solution you'd like**
We should name the types in a reproducible way based on the Data type definition, and not the variable name
For example, we could name arrays as `__array_<start>_<end>_<type>`
We could then reuse already defined types when we encounter them again.
Suggestions: 
```st
x : REF_TO INT; //__ref_int
x : INT(0..5); //__int_0_5
x : STRING[10]; //__string_10
x : ARRAY[0..1] OF INT(0..5); //__array_0_1_int_0_5
```
**Additional context**
Alias types might be harder to resolve, we should consider them here.

Once we have constants, these should probably be resolved first.

riederm, 2021-08-23:
We should prefix them in a way so you cannot create a derived type (alias) with a conflicting name.

riederm, 2021-11-08:
is this obsolete? or are we still aiming for this?

ghaith, 2021-11-09:
I'll close it for now.
We can try and tackle it another time (in another issue) if the need arises.

closed 2021-11-09 by ghaith