PLC-lang/rusty #273: Validation: Invalid Function return types

riederm, 2021-08-31:
As far as the parser is concerned, a Function can return a lot of different types - but not everything makes sense:

valid return types:
- Array-Types (ARRAY[0..3] OF INT, ...)
- Pointer-Types (POINTER TO INT, ...)
- String-Types (STRING[23], WSTRING[100], ...)
- Scalar Types (INT, BOOL, etc.)

invalid return types:
- Struct-Types ... _cannot be assigned because the type is unknown by any caller_
- Enum-Type ... _cannot be assigned because the type is unknown by any caller_
- VarArgs ... _makes no sense_

unclear ... 
- SubRangeType ... _maybe useful ... but will this be enforced by the compiler? i guess not?_

ghaith, 2021-09-01:
I assume the assignment of Subrange will be validated using the CHECK_XX method as every other variable. Not sure what else we could enforce there.

riederm, 2021-09-01:
> I assume the assignment of Subrange will be validated using the CHECK_XX method as every other variable. Not sure what else we could enforce there.

Good point ... So it could make sense to be used as a functions return