PLC-lang/rusty #286: basic compiler test failed

rarris, 2021-09-20:
let prog = "
    FUNCTION main : DINT
        main := (6 * 100) + (600 / 6) - 500 + (200 / 20) - 210;
    END_FUNCTION
    ";

this test should have the result: "0", but the compiler gives "400"

ghaith, 2021-09-20:
Generated order  : 
`(6 * 100) + ((600/6) - (500+((200/20) - 210)))`
Generated AST : 

```
statements: [
Assignment {
		left: Reference {
				name: "main",
		},
		right: BinaryExpression {
				operator: Plus,
				left: BinaryExpression {
						operator: Multiplication,
						left: LiteralInteger {
								value: 6,
						},
						right: LiteralInteger {
								value: 100,
						},
				},
				right: BinaryExpression {
						operator: Minus,
						left: BinaryExpression {
								operator: Division,
								left: LiteralInteger {
										value: 600,
								},
								right: LiteralInteger {
										value: 6,
								},
						},
						right: BinaryExpression {
								operator: Plus,
								left: LiteralInteger {
										value: 500,
								},
								right: BinaryExpression {
										operator: Minus,
										left: BinaryExpression {
												operator: Division,
												left: LiteralInteger {
														value: 200,
												},
												right: LiteralInteger {
														value: 20,
												},
										},
										right: LiteralInteger {
												value: 210,
										},
								},
						},
				},
		},
},
],

```
Clang Order : `((((((6*100))+(600/6))-500)+(200/20))-(210))`
Clang AST for comparison: 
```graphviz
        BinaryOperator 0x55839116d178 <col:12, col:53> 'int' '-'
        |-BinaryOperator 0x55839116d138 <col:12, col:49> 'int' '+'
        | |-BinaryOperator 0x55839116d098 <col:12, col:34> 'int' '-'
        | | |-BinaryOperator 0x55839116d058 <col:12, col:30> 'int' '+'
        | | | |-ParenExpr 0x55839116cfb8 <col:12, col:20> 'int'
        | | | | `-BinaryOperator 0x55839116cf98 <col:13, col:17> 'int' '*'
        | | | |   |-IntegerLiteral 0x55839116cf58 <col:13> 'int' 6
        | | | |   `-IntegerLiteral 0x55839116cf78 <col:17> 'int' 100
        | | | `-ParenExpr 0x55839116d038 <col:24, col:30> 'int'
        | | |   `-BinaryOperator 0x55839116d018 <col:25, col:29> 'int' '/'
        | | |     |-IntegerLiteral 0x55839116cfd8 <col:25> 'int' 600
        | | |     `-IntegerLiteral 0x55839116cff8 <col:29> 'int' 6
        | | `-IntegerLiteral 0x55839116d078 <col:34> 'int' 500
        | `-ParenExpr 0x55839116d118 <col:40, col:49> 'int'
        |   `-BinaryOperator 0x55839116d0f8 <col:41, col:47> 'int' '/'
        |     |-IntegerLiteral 0x55839116d0b8 <col:41> 'int' 200
        |     `-IntegerLiteral 0x55839116d0d8 <col:47> 'int' 20
        `-IntegerLiteral 0x55839116d158 <col:53> 'int' 210

```

riederm, 2021-09-20:
we treat `a - b + c` like `a - (b + c)`

closed 2021-09-22 by ghaith