PLC-lang/rusty #287: Store f64 in LiteralReal

riederm, 2021-09-20:
Literal reals are currently stored as a String inside the AstStatement::LiteralReal. This forces to reparse the value during constant-processing and also during validation.

Is there a reason why we store it as a String? It would be much more convinient to have this available as an f64 from the AST.

ghaith, 2021-09-20:
I think it's to preserve the exponents as they were written by the user

riederm, 2021-09-21:
> I think it's to preserve the exponents as they were written by the user

LiteralIntegers parse the value right away into an i128, also for hex, oct or bin values

ghaith, 2021-09-21:
Yes, I was thinking a bit more about it and I assume we could do that.
I'm not sure if a conversion from an f64 to an f32 could be as harmless as the i128 to i32 when the underlying type is only an f32

ghaith, 2022-06-22:
Some random guy on the internet said we can store f32 in f64s
https://stackoverflow.com/questions/59794757/are-all-32-bit-floats-exactly-represented-as-f64

closed 2022-12-07 by ghaith