PLC-lang/rusty #313: TIME mixed math calculation overflow

rarris, 2021-09-28:
let prog = "
    FUNCTION main : TIME
    VAR
        t1 : TIME := T#5s;
        t2 : TIME := T#6s;
        t3 : TIME := T#3s;
        t4 : TIME := T#2s;
        t5 : TIME := T#10s;
    END_VAR
        main := t1 + t2 * t3 / t4 - t5;
    END_FUNCTION
    ";

program result: -5223372036
expected result: 4000000000

ghaith, 2021-09-28:
I think the problem is that we save times in nanoseconds here, and this calculation overflows
The reason it overflows is that t2 * t3 is much bigger than 64bit
I also don't think that multiplying 5 seconds by 5 seconds is also normal as the unit here would be seconds squared, you usually would have multiplied by 5, not 5 seconds.
Same for division

rarris, 2021-09-28:
updated working test:

let prog = "
    FUNCTION main : TIME
    VAR
        t1 : TIME := T#5s;
        t2 : TIME := T#6s;
        t3 : TIME := T#10s;
    END_VAR
        main := t1 + t2 * 3 / 2 - t3;
    END_FUNCTION
    ";