PLC-lang/rusty #386: Can not compile hello_world.st

vitw, 2021-11-29:
I think I am missing something in here. 
I try to build hello_world.st with the following command:
`./rustyc -c examples/hello_world.st -o hello_world.o`

And get the following in terminal:

error: Unexpected token: expected Identifier but found exit
  ┌─ examples/hello_world.st:7:20
  │
7 │ @EXTERNAL FUNCTION exit : DINT
  │                    ^^^^

error: Unexpected token: expected Literal but found :
  ┌─ examples/hello_world.st:7:25
  │
7 │ @EXTERNAL FUNCTION exit : DINT
  │                         ^

error: Unexpected token: expected KeywordSemicolon but found 'VAR_INPUT
    status'
  ┌─ examples/hello_world.st:8:1
  │  
8 │ ╭ VAR_INPUT
9 │ │     status : DINT;
  │ ╰──────────^

error: Unexpected token: expected Literal but found END_VAR
   ┌─ examples/hello_world.st:10:1
   │
10 │ END_VAR
   │ ^^^^^^^

error: Unexpected token: expected KeywordSemicolon but found 'END_VAR'
   ┌─ examples/hello_world.st:10:1
   │
10 │ END_VAR
   │ ^^^^^^^

error: Missing expected Token [KeywordSemicolon, KeywordColon]
   ┌─ examples/hello_world.st:11:1
   │
11 │ END_FUNCTION
   │ ^^^^^^^^^^^^

error: Unexpected token: expected KeywordSemicolon but found 'END_FUNCTION'
   ┌─ examples/hello_world.st:11:1
   │
11 │ END_FUNCTION
   │ ^^^^^^^^^^^^

error: Function Return type missing
   ┌─ examples/hello_world.st:7:11
   │  
 7 │   @EXTERNAL FUNCTION exit : DINT
   │ ╭───────────^
 8 │ │ VAR_INPUT
 9 │ │     status : DINT;
10 │ │ END_VAR
11 │ │ END_FUNCTION
   │ ╰────────────^

error: Could not resolve reference to DINT
  ┌─ examples/hello_world.st:7:27
  │
7 │ @EXTERNAL FUNCTION exit : DINT
  │                           ^^^^

error: Could not resolve reference to DINT
  ┌─ examples/hello_world.st:9:14
  │
9 │     status : DINT;
  │              ^^^^

Error: Cannot break out of loop when not inside a loop

riederm, 2021-11-29:
I'm afraid the files in `ecamples/`are not up to date.
in the meantime we added `exit` as a keyword which now conflicts with a real executable with a `_start` entry point.

I think for the moment the main use-case for this compiler is to compile a library (an object) that is loaded and called by some osrt of runtime (e.g. a c-program).

@ghaith is there any other way to get a self-contained executable since we added `exit` as a keyword?
anyways we should update the examples forlder and add documentation on how to compile them

ghaith, 2021-11-29:
We could theoretically still compile an object with the main function (using the `-c` option) and link it against a crt0.o using lld, or simply call clang on the resulting object file.
issue #310 is the report for the initial problem.
Our strategies there would be either:
- remove case insensitive parsing (at least for `EXIT`)
- provide an own runtime object that contains the `_start` and `exit` functions which would call for an entry point (a well defined `main` function)

ghaith, 2022-12-07:
@riederm could you please check the files in the examples folder and also verify if the example's hello world builds

volsa, 2026-01-28:
Works fine on my end. We should probably revisit the examples at some point and tidy them up a bit (e.g. a Makefile would make sense?) but that is a low priority issue currently.