PLC-lang/rusty #391: Segfault during compilation

ghaith, 2021-12-02:
**Describe the bug**
Did not investigate in detail yet, but the following segfaults:

```st

TYPE movement : STRUCT 
	direction: DINT; 
	distance: DINT;
END_STRUCT
END_TYPE

VAR_GLOBAL CONSTANT
	forward : DINT := 1;
	up : DINT := 2;
	down : DINT := 3;
	len : DINT := 6;
END_VAR

FUNCTION main : DINT
	drive();
END_FUNCTION

FUNCTION drive : DINT
VAR
	//Not initialized here due to #362
	input : ARRAY[0..len-1] OF movement;
	horiz, depth : DINT;
END_VAR
VAR_TEMP
	counter : DINT := 0;
END_VAR

set_input(input := input);

FOR counter := 0 TO len - 1 DO
	CASE input[counter].direction OF
		forward : 
			horiz := horiz + input[counter].distance;
		up :
			depth := depth - input[counter].distance;
		down : 
			depth := depth + input[counter].distance;
	END_CASE
END_FOR

printf('Horiz : %d, Depth: %d, Result : %d', horiz, depth, depth * horiz);

END_FUNCTION

FUNCTION set_input : DINT
VAR_IN_OUT
	input : ARRAY[0..len-1] OF movement;
END_VAR
VAR_TEMP
	counter : DINT := 0;
END_VAR

// Vim command to convert the input : 
// %s/\v([a-z]+) (\d)/input[counter].diection := \1; input[counter].distance := \2; counter := counter + 1;/g

input[counter].direction := forward; input[counter].distance := 5; counter := counter + 1;
input[counter].direction := down; input[counter].distance := 5; counter := counter + 1;
input[counter].direction := forward; input[counter].distance := 8; counter := counter + 1;
input[counter].direction := up; input[counter].distance := 3; counter := counter + 1;
input[counter].direction := down; input[counter].distance := 8; counter := counter + 1;
input[counter].direction := forward; input[counter].distance := 2; counter := counter + 1;

END_FUNCTION

@EXTERNAL
FUNCTION printf : INT
VAR_INPUT
	str : STRING[1024];
	params: ...;
END_VAR
END_FUNCTION

```
Regardless of the status/correctness of the code, we should never segfault

ghaith, 2021-12-02:
The issue seems to be in the case statement

riederm, 2021-12-02:
it segfaults when executing? or during compilation?

ghaith, 2021-12-02:
During compilation :(

riederm, 2022-01-23:
the problem is in the case
```
	CASE input[counter].direction OF
		forward : 
			horiz := horiz + input[counter].distance;
		up :
			depth := depth - input[counter].distance;
		down : 
			depth := depth + input[counter].distance;
	END_CASE
```

I believe the problem lays in the `forward`, `up` and `down` constants. If you turn these into literals, the code compiles.

commit 01f0e530 (2022-06-19): constant propagation support

when generating an expression we check if this may refer to a constant
expression and rather generate it's initial value instead of a
reference to the global variable.

fixes #391 and #291
commit 85e75144 (2022-06-19): constant propagation support

when generating an expression we check if this may refer to a constant
expression and rather generate it's initial value instead of a
reference to the global variable.

fixes #391 and #291
commit 01632810 (2022-06-21): constant propagation support (#504)

* constant propagation support

when generating an expression we check if this may refer to a constant
expression and rather generate its initial value instead of a
reference to the global variable.

fixes #391 and #291

* added correctness tests for constant propagation

testcases:
- constants & const-expressions used as initial values
- constants & const-expressions used in case-statement
- constants & const-expressions used in array-declaration