PLC-lang/rusty #395: validate Pointer - Numerics assignments if it is no perfect match

riederm, 2021-12-02:
we support assigning between numbers and pointers in both directions because this somehow has a tradition in ST (and OSCAT). 
```st
PROGRAM FOO
   VAR
      ptr : POINTER TO INT;
      adress : LWORD;
   END_VAR
   
   LWORD := 16#DEAD_BEEF;
   ptr := LWORD;  //redirect the pointer to the adress
   ...
   LWORD := ptr;  //store the adress in a LWORD
   ...
END_PROGRAM
```

rusty simply casts between the number and the pointer-type. However if the number is smaller than the platform's pointer-size, the adress gets simply truncated.

The validator should warn you against such assignments (or should this event become an error?!)

riederm, 2021-12-02:
at the moment it is not clear how we determine how big the current platform's adress-size

ghaith, 2021-12-03:
I think we could use the target (So default platform target or the passed argument) to know what platform we target and then hold a map for the common ones.
We could also allow overrides through config

closed 2022-07-14 by ghaith