PLC-lang/rusty #402: Methods cannot resolve non class functions

ghaith, 2021-12-05:
**Describe the bug**
Given an external function and a class method, the validator will always report a reference issue with the function if called from that method

**To Reproduce**
```st
FUNCTION foo : DINT
END_FUNCTION

CLASS baz 
METHOD x : DINT
	foo();
END_METHOD
END_CLASS

```

r2k-in-the-vortex, 2022-11-27:
Someone probably already fixed this, unable to reproduce.

```
{external} 
FUNCTION puts : DINT
VAR_INPUT {ref}
    text : STRING;
END_VAR
END_FUNCTION

{external} 
FUNCTION sprintf : DINT
VAR_INPUT {ref}
    str     : STRING;
    format  : STRING;
END_VAR
VAR_INPUT
    i      : DINT;
END_VAR
END_FUNCTION

FUNCTION DINT_TO_STRING : STRING
    VAR_INPUT
        i      : DINT;
    END_VAR
    sprintf(DINT_TO_STRING, '%d', i);
END_FUNCTION

FUNCTION foo : DINT
    foo := 42;
END_FUNCTION

CLASS baz 
METHOD x : DINT
	x := foo();
END_METHOD
END_CLASS

FUNCTION main : DINT
    VAR
        Variable    : baz; (* Comment *)
        x           : DINT;
        test        : STRING;
    END_VAR
    
    x := Variable.x();
    test := DINT_TO_STRING(x);
    puts(test);
    puts('it works just fine');
END_FUNCTION

```
produces
```
vscode ➜ /workspaces/rusty $ target/release/rustyc examples/bug.st --linker=cc && ./bug
42
it works just fine
```

riederm, 2022-11-27:
thx @r2k-in-the-vortex for your efforts, probably you are right.
we halted the implementation for classes for the moment because we first want to fully support the pre 2013 extensions of the iec61131-3 standard. 

But it looks like we're very close to feature completeness (for <2013) so we will tackle all the class issues soon.

i'll look at it tomorrow, maybe I can find the commit that fixed this and link it, otherwise we can just close this.

r2k-in-the-vortex, 2022-11-28:
If root of the problem is unknown and so is the fix, does it require a testcase to be added so the problem wouldn't sneak back in sometime in future?

ghaith, 2022-11-28:
@r2k-in-the-vortex Yes I think we should have a test case covering it even if we are not yet officially supporting methods, I think @riederm can upload a commit if he is verifying this today, or I can add a test otherwise.

closed 2022-12-06 by 99NIMI