PLC-lang/rusty #411: Bring the coverage back to ~95%

ghaith, 2021-12-09:
**Is your refactor request related to a problem? Please describe.**
It looks like our coverage values dipped recently.
Looking at codecov results, there are some obvious candidates to improve 

**Describe the solution you'd like**
We can use this story as a tracker until we are back to around 95%

ghaith, 2022-06-22:
We should look at code cov,  and see if we have a problem in configuration

Some suspect files to improve : 
llvm_typesystem : casting errors and conversion of int to float
diagnostic : relocate and pragma
const_expressions
linker : add lib and in general not well covered
date_time_utils : invalid date

volsa, 2023-05-08:
Can we close this? The current coverage is almost at 97%

ghaith, 2023-05-08:
Indeed.

closed 2023-05-08 by ghaith