PLC-lang/rusty #417: Segfault when compiling oscat

ghaith, 2021-12-10:
**Describe the bug**
Compiling oscat fails with the following error (i have stopped pointer arithmetic in my version) : 
```
❯ ./rustyc -c *.st
error: Missing compare function 'FUNCTION __REPLACE_UML_pto_LESS : BOOL VAR_INPUT a,b : __REPLACE_UML_pto; END_VAR ...'.
      ┌─ oscat_single_file.st:25979:6
      │
25979 │     IF pto < ptm AND pt2^ > 0 THEN
      │        ^^^^^^^^^

zsh: segmentation fault (core dumped)  ./rustyc -c *.st

```
I generated the IR out of the lib and compiled it with LLC and got the following : 
```
oscat on  main [!?]
❯ llc oscat.ll -o oscat.o
llc: error: llc: oscat.ll:887:84: error: constant expression type mismatch: got type '[7 x i8]' but expected '[81 x i8]'
@LANGUAGE = global %CONSTANTS_LANGUAGE { i16 1, i16 3, [21 x [81 x i8]] [[81 x i8] c"Monday\00", [81 x i8] c"Tuesday\00", [81 x i8] c"Wednesday\00", [81 x i8] c"Thursday\00", [81 x i8] c"Friday\00", [81 x i8] c"Saturday\00", [81 x i8] c"Sunday\00", [81 x i8] c"Montag\00", [81 x i8] c"Dienstag\00", [81 x i8] c"Mittwoch\00", [81 x i8] c"Donnerstag\00", [81 x i8] c"Freitag\00", [81 x i8] c"Samstag\00", [81 x i8] c"Sonntag\00", [81 x i8] c"Lundi\00", [81 x i8] c"Mardi\00", [81 x i8] c"Mercredi\00", [81 x i8] c"Jeudi\00", [81 x i8] c"Vendredi\00", [81 x i8] c"Samedi\00", [81 x i8] c"Dimanche\00"], [21 x [81 x i8]] [[81 x i8] c"Mo\00", [81 x i8] c"Tu\00", [81 x i8] c"We\00", [81 x i8] c"Th\00", [81 x i8] c"Fr\00", [81 x i8] c"Sa\00", [81 x i8] c"Su\00", [81 x i8] c"Mo\00", [81 x i8] c"Di\00", [81 x i8] c"Mi\00", [81 x i8] c"Do\00", [81 x i8] c"Fr\00", [81 x i8] c"Sa\00", [81 x i8] c"So\00", [81 x i8] c"Lu\00", [81 x i8] c"Ma\00", [81 x i8] c"Me\00", [81 x i8] c"Je\00", [81 x i8] c"Ve\00", [81 x i8] c"Sa\00", [81 x i8] c"Di\00"], [36 x [81 x i8]] [[81 x i8] c"January\00", [81 x i8] c"February\00", [81 x i8] c"March\00", [81 x i8] c"April\00", [81 x i8] c"May\00", [81 x i8] c"June\00", [81 x i8] c"July\00", [81 x i8] c"August\00", [81 x i8] c"September\00", [81 x i8] c"October\00", [81 x i8] c"November\00", [81 x i8] c"December\00", [81 x i8] c"Januar\00", [81 x i8] c"Februar\00", [81 x i8] c"M\EF\BF\BDrz\00", [81 x i8] c"April\00", [81 x i8] c"Mai\00", [81 x i8] c"Juni\00", [81 x i8] c"Juli\00", [81 x i8] c"August\00", [81 x i8] c"September\00", [81 x i8] c"Oktober\00", [81 x i8] c"November\00", [81 x i8] c"Dezember\00", [81 x i8] c"Janvier\00", [81 x i8] c"F\EF\BF\BDvrier\00", [81 x i8] c"mars\00", [81 x i8] c"Avril\00", [81 x i8] c"Mai\00", [81 x i8] c"Juin\00", [81 x i8] c"Juillet\00", [81 x i8] c"Ao\EF\BF\BDt\00", [81 x i8] c"Septembre\00", [81 x i8] c"Octobre\00", [81 x i8] c"Novembre\00", [81 x i8] c"Decembre\00"], [36 x [81 x i8]] [[81 x i8] c"Jan\00", [81 x i8] c"Feb\00", [81 x i8] c"Mar\00", [81 x i8] c"Apr\00", [81 x i8] c"May\00", [81 x i8] c"Jun\00", [81 x i8] c"Jul\00", [81 x i8] c"Aug\00", [81 x i8] c"Sep\00", [81 x i8] c"Oct\00", [81 x i8] c"Nov\00", [81 x i8] c"Dec\00", [81 x i8] c"Jan\00", [81 x i8] c"Feb\00", [81 x i8] c"Mrz\00", [81 x i8] c"Apr\00", [81 x i8] c"Mai\00", [81 x i8] c"Jun\00", [81 x i8] c"Jul\00", [81 x i8] c"Aug\00", [81 x i8] c"Sep\00", [81 x i8] c"Okt\00", [81 x i8] c"Nov\00", [81 x i8] c"Dez\00", [81 x i8] c"Jan\00", [81 x i8] c"Fev\00", [81 x i8] c"Mar\00", [81 x i8] c"Avr\00", [81 x i8] c"Mai\00", [81 x i8] c"Jun\00", [81 x i8] c"Jul\00", [81 x i8] c"Aou\00", [81 x i8] c"Sep\00", [81 x i8] c"Oct\00", [81 x i8] c"Nov\00", [81 x i8] c"Dec\00"], [48 x [81 x i8]] [[81 x i8] c"N\00", [81 x i8] c"NNE\00", [81 x i8] c"NE\00", [81 x i8] c"ENE\00", [81 x i8] c"E\00", [81 x i8] c"ESE\00", [81 x i8] c"SE\00", [81 x i8] c"SSE\00", [81 x i8] c"S\00", [81 x i8] c"SSW\00", [81 x i8] c"SW\00", [81 x i8] c"WSW\00", [81 x i8] c"W\00", [81 x i8] c"WNW\00", [81 x i8] c"NW\00", [81 x i8] c"NNW\00", [81 x i8] c"N\00", [81 x i8] c"NNO\00", [81 x i8] c"NO\00", [81 x i8] c"ONO\00", [81 x i8] c"O\00", [81 x i8] c"OSO\00", [81 x i8] c"SO\00", [81 x i8] c"SSO\00", [81 x i8] c"S\00", [81 x i8] c"SSW\00", [81 x i8] c"SW\00", [81 x i8] c"WSW\00", [81 x i8] c"W\00", [81 x i8] c"WNW\00", [81 x i8] c"NW\00", [81 x i8] c"NNW\00", [81 x i8] c"N\00", [81 x i8] c"NNO\00", [81 x i8] c"NO\00", [81 x i8] c"ONO\00", [81 x i8] c"O\00", [81 x i8] c"OSO\00", [81 x i8] c"SO\00", [81 x i8] c"SSO\00", [81 x i8] c"S\00", [81 x i8] c"SSW\00", [81 x i8] c"SW\00", [81 x i8] c"WSW\00", [81 x i8] c"W\00", [81 x i8] c"WNW\00", [81 x i8] c"NW\00", [81 x i8] c"NNW\00"] }
                                                                        
```

The relevant section from oscat is : 
```st
(* @NESTEDCOMMENTS := 'Yes' *)
(* @PATH := '' *)
(* @OBJECTFLAGS := '0, 8' *)
TYPE CONSTANTS_LANGUAGE :
STRUCT
	(* Language Setup *)
	DEFAULT : INT := 1; (* 1=english, 2=german 3=french *)
	LMAX : INT := 3;
	WEEKDAYS : ARRAY[1..3, 1..7] OF STRING[10] :=	['Monday', 'Tuesday', 'Wednesday', 'Thursday', 'Friday', 'Saturday', 'Sunday',
													'Montag', 'Dienstag', 'Mittwoch', 'Donnerstag', 'Freitag', 'Samstag', 'Sonntag',
													'Lundi', 'Mardi', 'Mercredi', 'Jeudi', 'Vendredi', 'Samedi', 'Dimanche'];
	WEEKDAYS2 : ARRAY[1..3, 1..7] OF STRING[2] :=	['Mo', 'Tu', 'We', 'Th', 'Fr', 'Sa', 'Su',
													'Mo', 'Di', 'Mi', 'Do', 'Fr', 'Sa', 'So',
													'Lu', 'Ma', 'Me', 'Je', 'Ve', 'Sa', 'Di'];
	MONTHS : ARRAY[1..3, 1..12] OF STRING[10] :=	['January', 'February', 'March', 'April', 'May', 'June', 'July', 'August', 'September', 'October', 'November', 'December',
													'Januar', 'Februar', 'M�rz', 'April', 'Mai', 'Juni', 'Juli', 'August', 'September', 'Oktober', 'November', 'Dezember',
													'Janvier', 'F�vrier', 'mars', 'Avril', 'Mai', 'Juin', 'Juillet', 'Ao�t', 'Septembre', 'Octobre', 'Novembre', 'Decembre'];
	MONTHS3 : ARRAY[1..3, 1..12] OF STRING[3] :=	['Jan', 'Feb', 'Mar', 'Apr', 'May', 'Jun', 'Jul', 'Aug', 'Sep', 'Oct', 'Nov', 'Dec',
													'Jan', 'Feb', 'Mrz', 'Apr', 'Mai', 'Jun', 'Jul', 'Aug', 'Sep', 'Okt', 'Nov', 'Dez',
													'Jan', 'Fev', 'Mar', 'Avr', 'Mai', 'Jun', 'Jul', 'Aou', 'Sep', 'Oct', 'Nov', 'Dec'];
	DIRS : ARRAY[1..3,0..15] OF STRING[3] :=		['N', 'NNE', 'NE', 'ENE', 'E', 'ESE', 'SE', 'SSE', 'S', 'SSW', 'SW', 'WSW', 'W', 'WNW', 'NW', 'NNW',
													'N', 'NNO', 'NO', 'ONO', 'O', 'OSO', 'SO', 'SSO', 'S', 'SSW', 'SW', 'WSW', 'W', 'WNW', 'NW', 'NNW',
													'N', 'NNO', 'NO', 'ONO', 'O', 'OSO', 'SO', 'SSO', 'S', 'SSW', 'SW', 'WSW', 'W', 'WNW', 'NW', 'NNW'];
END_STRUCT
END_TYPE
```

riederm, 2021-12-11:
does this mean that if we declare [81 x i8], we need to make sure to really provide 81 chars in initializiation?
like `c"Hello World\00\00\00...\00"`?

ghaith, 2021-12-11:
I think it's that we are assigning 81 sized strings to 7 sized strings, although they fit.
I could be wrong though, and I didn't dive deep into that code yet but maybe we should give the correct size hint /size to the literals in initial values here?

riederm, 2022-01-19:
Additional issues concerning building oscat are:
- #436 
- #435

PR #421 (2022-01-19): annotate and generate STRING-literals precisly according to their length
the annotator tells us exactly how long every string-literal is by craeting dedicated STRING-types for them. The expression generator now applies two different strategies:
- when generating string-literals *in declarations* (outside of a body) we generate the full length of the string-literal. So a `STRING[80] = 'a'` will generate 80 additional `\00` to fill the full vector
- when generating string-literals *in the body* we generate only the string's real length. So an assignment: `x := 'a';` will generate an vec<i8> of l [... cut]
commit 163a0c69 (2022-01-19): annotate and generate STRING-literals precisly according to their length (#421)

he annotator tells us exactly how long every string-literal is by craeting dedicated STRING-types for them. The expression generator now applies two different strategies:

when generating string-literals in declarations (outside of a body) we generate the full length of the string-literal. So a STRING[80] = 'a' will generate 80 additional \00 to fill the full vector
when generating string-literals in the body we generate only the string's real length. So an assignment: x := 'a'; will generate an vec of length 2 wi [... cut]