PLC-lang/rusty #420: Boolean values in arrays seem to leak into next elements

ghaith, 2021-12-13:
**Describe the bug**
Given a boolean array, when assigning a value for an element, it seems to affect followup elements

I'll add a clearer example later, but it's mostly something like
```
arr : ARRAY[0..2] OF BOOL := [3(FALSE)];
..
arr[1] := TRUE;
arr[0] := FALSE;

---> arr[1] will be false again
```

riederm, 2022-01-15:
I believe that we have a conceptual issue here since we use `u1` for bool.
I think the store statement has to write the whole byte. I dont think that the adress returned by the gep-expression can point to a single bit.

so we have 2 ways here:
1. we make booleans u8 like most languages do (although the norm states it wants a bit). I think this is better for performance.
2. we generate loads and stores to bool-arrays as bit-access operations. So for `boolArray[3]` we generate something like `(boolArray > 3) & 0x1` and load this as u1

riederm, 2022-01-15:
this does not load the 2nd bit! it loads the 2nd byte from bArray
```ll
%tmpVar1 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 1
```

this test works!!! althoug it should not
```rs
#[test]
fn bool_array_assignments() {
    #[repr(C)]
    struct MainType {
        y: i16,
        b_array: [u8; 8],   // i reserve 8 bytes here! instead of 1
        z: i16,
    }

    let function = r"
        PROGRAM main
        VAR
            y : INT;
            bArray : ARRAY[0..7] OF BOOL := [8(FALSE)];
            z : INT;
        END_VAR
            y := 111;
            z := 222;
            bArray[0] := TRUE;
            bArray[1] := FALSE;
            bArray[2] := TRUE;
            bArray[3] := FALSE;
            bArray[4] := TRUE;
            bArray[5] := FALSE;
            bArray[6] := TRUE;
            bArray[7] := FALSE;
        END_PROGRAM
        ";

    let mut maintype = MainType{
        y: 0, b_array: [0;8], z: 0
    };

    let _: i32 = compile_and_run(function.to_string(), &mut maintype);

    
    //here we see that the assignments went on byte-n instead of bit-n
    assert_eq!(maintype.b_array, [1,0,1,0,1,0,1,0]);
    
    assert_eq!(maintype.y, 111);
    assert_eq!(maintype.z, 222);
}
```
the resulting IR code uses `i1` but the test clearly shows that the semantic is different. The addresses stored in `%tempVar` cannot point to a certain bit.
```ll
%main_interface = type { i16, [8 x i1], i16 }

@main_instance = global %main_interface zeroinitializer

define void @main(%main_interface* %0) {
entry:
  %y = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 0
  %bArray = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 1
  %z = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 2
  store i16 111, i16* %y, align 2
  store i16 222, i16* %z, align 2
  %tmpVar = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 0
  store i1 true, i1* %tmpVar, align 1
  %tmpVar1 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 1
  store i1 false, i1* %tmpVar1, align 1
  %tmpVar2 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 2
  store i1 true, i1* %tmpVar2, align 1
  %tmpVar3 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 3
  store i1 false, i1* %tmpVar3, align 1
  %tmpVar4 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 4
  store i1 true, i1* %tmpVar4, align 1
  %tmpVar5 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 5
  store i1 false, i1* %tmpVar5, align 1
  %tmpVar6 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 6
  store i1 true, i1* %tmpVar6, align 1
  %tmpVar7 = getelementptr inbounds [8 x i1], [8 x i1]* %bArray, i32 0, i32 7
  store i1 false, i1* %tmpVar7, align 1
  ret void
}
```

riederm, 2022-01-16:
interesting experiment:
when i turn rusty's `BOOL` into  i8, all correctness-tests keep working except `correctness::bitaccess::bitaccess_assignment_should_not_override_current_values` ... it again overwrites some bits

ghaith, 2022-01-17:
> I believe that we have a conceptual issue here since we use `u1` for bool. I think the store statement has to write the whole byte. I dont think that the adress returned by the gep-expression can point to a single bit.
> 
> so we have 2 ways here:
> 
>     1. we make booleans u8 like most languages do (although the norm states it wants a bit). I think this is better for performance.
> 
>     2. we generate loads and stores to bool-arrays as bit-access operations. So for `boolArray[3]` we generate something like `(boolArray >> 3) & 0x1` and load this as u1
> 
> 
> opinions @ghaith ?

I think option 1 sounds better, but I did change to u1s for bitaccess.
We could look into the failing tests and see if we can change something on bit access.
Were you able to reproduce the initial behaviour? @99NIMI was not. I only had it during the AOC exercises, and I need to lookup the issue / code again for it.

riederm, 2022-02-08:
this was merged in #444

commit 2476ff92 (2022-01-26): turn bool into i8

generating bool as i1 makes it impossible to have bool-arrays
because an array-access (x[2]) can only adress (read/write) a
whole byte, not a single bit.

bit-access relies on bool beeing a single bit so I introduced an
internal (hidden) type called '__U1' that is really one bit.
If we detect a bool in bit-access, we internally switch form
BOOL to __U1 to get the original behavior

fixes #420
closed 2022-02-08 by riederm