PLC-lang/rusty #423: empty STRUCTs should be allowed

rarris, 2021-12-16:
**Describe the bug**
rusty compiler interprets an empty STRUCT as not ok, as **error**
in PLC Dev empty STRUCTs are allowed, but with a warning: "_The structure 'MV7000_Parameters_TYPE' does not contain any elements_"

**Additional context**
```
TYPE MV7000_Parameters_TYPE :
STRUCT
END_STRUCT
END_TYPE
```

ghaith, 2022-06-22:
We need a configurable diagnostic accessor

riederm, 2022-11-07:
I'm not sure how we handle this.
like can i declare such a variable?

```st
VAR_GLOBAL
   x : MV7000_Parameters_TYPE;
END_VAR
```
now is there actually some memory allocated? can I have an array of this type? can i do an adress-of operation `ADR(x)`? Is this now a 0-byte sized type? Or do we need to add a phantom filed with 1 byte to it? It confuses me.