PLC-lang/rusty #448: initializing return-variables introduces redundant stores

riederm, 2022-01-31:
return values are initialized by default. Most of the type you will write into the variable, so the initial memset was useless.

```st
FUNCTION foo : STRING
  foo := 'hello';
END_PROGRAM
```

generates:
```ll
 0 │ ; ModuleID = 'main'
 1 │ source_filename = "main"
 2 │ 
 3 │ %foo_interface = type {}
 4 │ 
 5 │ @utf08_literal_0 = unnamed_addr constant [6 x i8] c"hello\00"
 6 │ 
 7 │ define [81 x i8] @foo(%foo_interface* %0) {
 8 │ entry:
 9 │   %foo = alloca [81 x i8], align 1
10 │   %1 = bitcast [81 x i8]* %foo to i8*
11 │   call void @llvm.memset.p0i8.i64(i8* align 1 %1, i8 0, i64 mul nuw (i64 ptrtoint (i8* getelementptr (i8, i8* null, i32 1) to i64), i64 81), i1 false)
12 │   %2 = bitcast [81 x i8]* %foo to i8*
13 │   call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %2, i8* align 1 getelementptr inbounds ([6 x i8], [6 x i8]* @utf08_literal_0, i32 0, i32 0), i32 6, i1 false)
14 │   %foo_ret = load [81 x i8], [81 x i8]* %foo, align 1
15 │   ret [81 x i8] %foo_ret
16 │ }
        ...
```
here line 10 and 11 initialize the string to `0`, line 12 and 13 immediately re-write it.
I'd like to verify that the llvm pass have no problem to get rid of the inital memset, also fo for more complex situations.

ghaith, 2022-06-22:
This needs flow analysis to remove these kind of assignment before codegen (Not only returns)

ghaith, 2026-02-09:
this seems to be fixed, there's no extra memset anymore

closed 2026-02-09 by ghaith