PLC-lang/rusty #450: Can't compile in Fedora 25

hiperiondev, 2022-02-01:
cargo build --release

breaks with:
[...]
 Compiling thiserror v1.0.30
   Compiling lld_rs v120.0.0 (https://github.com/ghaith/lld-rs?branch=main#8cd17040)
   Compiling inkwell v0.1.0 (https://github.com/TheDan64/inkwell?branch=master#d44ec8be)
error: could not find native static library `lldCOFF`, perhaps an -L flag is missing?

error: could not compile `lld_rs` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed

ghaith, 2022-02-01:
Hello,

you might need to install the dev libraries for LLVM/Clang and LLD
I don't currently have a fedora install so I was not able to verify it, but I will try it on a VM later today.
In the meantime, you could also use the docker images (Debian based) to get started. 
https://github.com/PLC-lang/rust-llvm-images
I use the linux image locally with podman and docker (For Dev containers)

riederm, 2022-02-01:
but beware, we still have some outdated test-files in the /examples folder which do not successfully compile

we should finally schedule #386

ghaith, 2022-02-01:
I experimented a bit on a fedora 35 VM and here's what I found
The default clang version is 13, we use 12 which is currently a blocker for the linking, if I disable the linker code (Remove it from cargo and comment out the link method in lib.rs) I mange to get further
I had to then install `libffi-devel` and `zlib-devel` (in addition to `clang-devel`, `llvm-devel`, `lld-devel`
This got me to compile once inkwell was on version 13 instead of 12.

I'm currently trying to compile an llvm-12 version on fedora to run with it. I'll update the post with that information.

hiperiondev, 2022-02-01:
In fedora exist llvm12 and can be installed with dnf

ghaith, 2022-02-02:
> In fedora exist llvm12 and can be installed via dnf

I am not familiar with dnf, but a dnf search only showed me clang with version 13. I did find clang12-devel, clang12-libs as well as llvm12-devel and llvm12-libs but these are just development libraries.
If you can provide the installation instructions I can try this further.
For now all I have is a guess that the binaries from lld and lld-devel distributed on fedora do not include the COFF lib by default, which is why I tried to compile on my own but it kept crashing on my VM, I'll try that again at a later point this week.

hiperiondev, 2022-02-10:
I have downloaded precompiled LLVM 12 (https://github.com/llvm/llvm-project/releases/download/llvmorg-12.0.1/clang+llvm-12.0.1-x86_64-linux-gnu-ubuntu-16.04.tar.xz) and seems to work.
But I don't know how to indicate to cargo to use this libraries

hiperiondev, 2022-02-14:
I have followed your recommendations but without success. Same error

ghaith, 2022-02-15:
Hello,
i just experimented with removing the COFF (and other dependencies) from the linker but I can't test on fedora
Could you try to do the following change in ruSTy's `Cargo.toml` and see if it builds on fedora with the default toolchain?

Change : `lld_rs = { git = "https://github.com/ghaith/lld-rs", branch = "main" }` To :
`lld_rs = { git = "https://github.com/ghaith/lld-rs", branch = "no_coff" }`

hiperiondev, 2022-02-15:
I have tested the no_coff alternative and now there is a different error:

Compiling lld_rs v120.0.0 (https://github.com/ghaith/lld-rs?branch=no_coff#c21c57b8)
error: could not find native static library `lldCommon`, perhaps an -L flag is missing?

error: could not compile `lld_rs` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed

I have  lld-devel and lld-libs installed:
/usr/lib64/liblldCommon.so
/usr/lib64/liblldCommon.so.13

ghaith, 2022-02-20:
> I have tested the no_coff alternative and now there is a different error:
>
> Compiling lld_rs v120.0.0 (https://github.com/ghaith/lld-rs?branch=no_coff#c21c57b8) error: could not find native static [...]

We depend on the static versions of lld as we link everything into one executable.
We were considering the options of using shared/installed llvm/lld/clang versions to do the linking but we still didn't get that far (help always welcome 😃).

I did some experimentation however with building clang and lld on fedora and was able to get ruSTy building with my local version
I have the steps in the following docker file, but you can just take the steps out and run them locally:

```Dockerfile
FROM fedora:latest as builder

# GCC kept crashing when building llvm, so we used clang.
# If you want a clean environment use a multistage docker to remove the initial clang
RUN dnf install -y git clang lld gcc g++ cmake python

# Clone only the needed branch. 
RUN git clone --depth 1 https://github.com/llvm/llvm-project.git -b llvmorg-12.0.1 /opt/llvm-project

RUN mkdir -p /opt/build
WORKDIR /opt/build

RUN cmake -DLLVM_ENABLE_LLD=true -DLLVM_ENABLE_PROJECTS="clang;lld" -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release /opt/llvm-project/llvm

RUN make -j 9

# Install clang/llvm into /opt/llvm 
RUN mkdir -p /opt/llvm
RUN cmake -DCMAKE_INSTALL_PREFIX=/opt/llvm -P cmake_install.cmake

# Multi state to keep the run system clean
FROM fedora:latest as runner

COPY --from=builder /opt/llvm /opt/llvm

ENV PATH="/opt/llvm/bin:${PATH}"

# Install Local depedencies for the runner
RUN dnf install -y git rust cargo libffi-devel gcc g++ 

RUN mkdir -p /opt/ws

WORKDIR /opt/ws
```

ghaith, 2022-04-26:
I will close the issue for now since there has been no update, the docker file to build LLVM should be fine, we also updated to LLVM13 in the meantime.

If you still have issues running on fedora feel free to comment on this issue and I will reopen it.

closed 2022-04-26 by ghaith