PLC-lang/rusty #468: Function calls should pass single args instead of POU-like instance struct

riederm, 2022-02-17:
**Describe the bug**
as described in https://plc-lang.github.io/rusty/arch/codegen.html#programs a call to a function also passes one single parameter-struct with all parameters, instead of the single paramters. This makes it very hard to include external c-functions since they expect all of their parameters on the stack.

**Expected behavior**
we should generate a traditional call for functions instead
so in pseudo-code we should generate .. 
```
foo(1, 2, 3, 4);
```
... instead of ... 
```
foo_instance f;
f.p1 = 1;
f.p2 = 2;
f.p3 = 3;
f.p4 = 4;
foo(f);
```

programs and function-blocks still should use an instance-struct for the parameres since they are persistent over calls.

riederm, 2022-03-23:
What needs to be done:

- [ ] a Function pou must not generate a state-struct-type
- [ ] a function's method must take all parameters, not the struct-type
- [ ] calling a function passes all parameters, not a single state-struct
- [ ] calling a function passes all parameters as pointers

ghaith, 2022-06-15:
I  think we could close this now that we support {ref} properties for pointers.

closed 2022-06-22 by ghaith