PLC-lang/rusty #477: Create a Plain text diagnostics reporter

ghaith, 2022-03-30:
**Is your feature request related to a problem? Please describe.**
The error reporter currently produces a rich format that cannot be displayed correctly on all terminals (Eclipse Console cannot display the symbols/colors correctly)
In addition, the output is hard for a tool to parse, for example, to display the errors inside an editor.

**Describe the solution you'd like**

- [ ] Introduce an `error-format` flag with the default being set to `rich` (our current output)
- [ ] Introduce a `text` format that displays each diagnostic on a single line, with enough information to also parse the diagnostic.

Suggested format : 

- Type : E/W/I 
- Path : Path to file
- Line : Start line of the diagnostic, optional
- Column: Start column of the diagnostic in the provided line, optional
- Range: Start and end offset of the diagnostic in the file, optional

```
<Type> <Path>[:<Line>[:<Column>]] [<Range>] <Message>
```
Example : 
```
E src/file.st:5:6 (10..50) Cannot resolve reference to `MyVar`
W src/file2.st [50..60]  `POINTER_TO` should be replaced with `REF_TO`
I src/file2.st Some information messages without a range
```

**Describe alternatives you've considered**
- Other easier to parse formats could work here (Like JSON maybe), but text should be enough for now.

riederm, 2022-03-30:
clang doc
https://releases.llvm.org/3.6.2/tools/docs/UsersManual.html#formatting-of-diagnostics

gcc doc
https://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Message-Formatting-Options.html

lsp:
https://microsoft.github.io/language-server-protocol/specifications/specification-current/#textDocument_publishDiagnostics

ghaith, 2022-04-04:
Updated clang manual 
https://releases.llvm.org/13.0.0/tools/clang/docs/UsersManual.html#formatting-of-diagnostics

closed 2022-04-26 by ghaith