PLC-lang/rusty #495: Strings seem to be misbehaving

ghaith, 2022-05-29:
**Describe the bug**
The following PR tries to update the Standard functions conversion functions to the new call style.
https://github.com/PLC-lang/StandardFunctions/pull/33
I could have missed something in the functions themselves, but is seems like the memcpy before a string call didn't copy the strings as expected : 

```ll
%main_interface = type { [81 x i8], [81 x i8]* }

@main_instance = global %main_interface zeroinitializer

define void @main(%main_interface* %0) {
entry:
  %res = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 0
  %ptr = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 1
  %1 = alloca [81 x i16], align 2
  %2 = bitcast [81 x i16]* %1 to i8*
  call void @llvm.memset.p0i8.i64(i8* align 1 %2, i8 0, i64 ptrtoint ([81 x i16]* getelementptr ([81 x i16], [81 x i16]* null, i32 1) to i64), i1 false)
  %3 = alloca [6 x i16], align 2
  store [6 x i16] [i16 104, i16 101, i16 108, i16 108, i16 111, i16 0], [6 x i16]* %3, align 2
  %4 = bitcast [81 x i16]* %1 to i8*
  %5 = bitcast [6 x i16]* %3 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %4, i8* align 1 %5, i32 80, i1 false)
  %6 = load [81 x i16], [81 x i16]* %1, align 2
  %call = call [81 x i8] @WSTRING_TO_STRING([81 x i16] %6)
  %7 = alloca [81 x i8], align 1
  store [81 x i8] %call, [81 x i8]* %7, align 1
  %8 = bitcast [81 x i8]* %res to i8*
  %9 = bitcast [81 x i8]* %7 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %8, i8* align 1 %9, i32 80, i1 false)
  ret void

entry1:                                           ; No predecessors!
  %res2 = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 0
  %ptr3 = getelementptr inbounds %main_interface, %main_interface* %0, i32 0, i32 1
  %10 = alloca [81 x i16], align 2
  %11 = bitcast [81 x i16]* %10 to i8*
  call void @llvm.memset.p0i8.i64(i8* align 1 %11, i8 0, i64 ptrtoint ([81 x i16]* getelementptr ([81 x i16], [81 x i16]* null, i32 1) to i64), i1 false)
  %12 = alloca [6 x i16], align 2
  store [6 x i16] [i16 104, i16 101, i16 108, i16 108, i16 111, i16 0], [6 x i16]* %12, align 2
  %13 = bitcast [81 x i16]* %10 to i8*
  %14 = bitcast [6 x i16]* %12 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %13, i8* align 1 %14, i32 80, i1 false)
  %15 = load [81 x i16], [81 x i16]* %10, align 2
  %call4 = call [81 x i8] @WSTRING_TO_STRING([81 x i16] %15)
  %16 = alloca [81 x i8], align 1
  store [81 x i8] %call4, [81 x i8]* %16, align 1
  %17 = bitcast [81 x i8]* %res2 to i8*
  %18 = bitcast [81 x i8]* %16 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %17, i8* align 1 %18, i32 80, i1 false)
  ret void
}

declare [81 x i8] @WSTRING_TO_STRING([81 x i16])

define i16 @WSTRING_TO_WCHAR([81 x i16] %0) {
entry:
  %in = alloca [81 x i16], align 2
  store [81 x i16] %0, [81 x i16]* %in, align 2
  %ptr = alloca i16*, align 8
  %WSTRING_TO_WCHAR = alloca i16, align 2
  store i16* null, i16** %ptr, align 8
  store i16 0, i16* %WSTRING_TO_WCHAR, align 2
  %1 = bitcast [81 x i16]* %in to i16*
  store i16* %1, i16** %ptr, align 8
  %deref = load i16*, i16** %ptr, align 8
  %load_tmpVar = load i16, i16* %deref, align 2
  store i16 %load_tmpVar, i16* %WSTRING_TO_WCHAR, align 2
  %WSTRING_TO_WCHAR_ret = load i16, i16* %WSTRING_TO_WCHAR, align 2
  ret i16 %WSTRING_TO_WCHAR_ret
}

declare [81 x i16] @STRING_TO_WSTRING([81 x i8])

define i8 @STRING_TO_CHAR([81 x i8] %0) {
entry:
  %in = alloca [81 x i8], align 1
  store [81 x i8] %0, [81 x i8]* %in, align 1
  %ptr = alloca i8*, align 8
  %STRING_TO_CHAR = alloca i8, align 1
  store i8* null, i8** %ptr, align 8
  store i8 0, i8* %STRING_TO_CHAR, align 1
  %1 = bitcast [81 x i8]* %in to i8*
  store i8* %1, i8** %ptr, align 8
  %deref = load i8*, i8** %ptr, align 8
  %load_tmpVar = load i8, i8* %deref, align 1
  store i8 %load_tmpVar, i8* %STRING_TO_CHAR, align 1
  %STRING_TO_CHAR_ret = load i8, i8* %STRING_TO_CHAR, align 1
  ret i8 %STRING_TO_CHAR_ret
}

define [81 x i16] @WCHAR_TO_WSTRING(i16 %0) {
entry:
  %in = alloca i16, align 2
  store i16 %0, i16* %in, align 2
  %ptr = alloca [81 x i16]*, align 8
  %WCHAR_TO_WSTRING = alloca [81 x i16], align 2
  store [81 x i16]* null, [81 x i16]** %ptr, align 8
  %1 = bitcast [81 x i16]* %WCHAR_TO_WSTRING to i8*
  call void @llvm.memset.p0i8.i64(i8* align 1 %1, i8 0, i64 ptrtoint ([81 x i16]* getelementptr ([81 x i16], [81 x i16]* null, i32 1) to i64), i1 false)
  %2 = bitcast i16* %in to [81 x i16]*
  store [81 x i16]* %2, [81 x i16]** %ptr, align 8
  %deref = load [81 x i16]*, [81 x i16]** %ptr, align 8
  %load_tmpVar = load [81 x i16], [81 x i16]* %deref, align 2
  %3 = alloca [81 x i16], align 2
  store [81 x i16] %load_tmpVar, [81 x i16]* %3, align 2
  %4 = bitcast [81 x i16]* %WCHAR_TO_WSTRING to i8*
  %5 = bitcast [81 x i16]* %3 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %4, i8* align 1 %5, i32 80, i1 false)
  %WCHAR_TO_WSTRING_ret = load [81 x i16], [81 x i16]* %WCHAR_TO_WSTRING, align 2
  ret [81 x i16] %WCHAR_TO_WSTRING_ret
}

declare i8 @WCHAR_TO_CHAR(i16)

define [81 x i8] @CHAR_TO_STRING(i8 %0) {
entry:
  %in = alloca i8, align 1
  store i8 %0, i8* %in, align 1
  %ptr = alloca [81 x i8]*, align 8
  %CHAR_TO_STRING = alloca [81 x i8], align 1
  store [81 x i8]* null, [81 x i8]** %ptr, align 8
  %1 = bitcast [81 x i8]* %CHAR_TO_STRING to i8*
  call void @llvm.memset.p0i8.i64(i8* align 1 %1, i8 0, i64 ptrtoint ([81 x i8]* getelementptr ([81 x i8], [81 x i8]* null, i32 1) to i64), i1 false)
  %2 = bitcast i8* %in to [81 x i8]*
  store [81 x i8]* %2, [81 x i8]** %ptr, align 8
  %deref = load [81 x i8]*, [81 x i8]** %ptr, align 8
  %load_tmpVar = load [81 x i8], [81 x i8]* %deref, align 1
  %3 = alloca [81 x i8], align 1
  store [81 x i8] %load_tmpVar, [81 x i8]* %3, align 1
  %4 = bitcast [81 x i8]* %CHAR_TO_STRING to i8*
  %5 = bitcast [81 x i8]* %3 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %4, i8* align 1 %5, i32 80, i1 false)
  %CHAR_TO_STRING_ret = load [81 x i8], [81 x i8]* %CHAR_TO_STRING, align 1
  ret [81 x i8] %CHAR_TO_STRING_ret
}

declare i16 @CHAR_TO_WCHAR(i8)

; Function Attrs: argmemonly nofree nounwind willreturn writeonly
declare void @llvm.memset.p0i8.i64(i8* nocapture writeonly, i8, i64, i1 immarg) #0

; Function Attrs: argmemonly nofree nounwind willreturn
declare void @llvm.memcpy.p0i8.p0i8.i32(i8* noalias nocapture writeonly, i8* noalias nocapture readonly, i32, i1 immarg) #1

attributes #0 = { argmemonly nofree nounwind willreturn writeonly }
attributes #1 = { argmemonly nofree nounwind willreturn }
"

```

riederm, 2022-05-30:
it looks like the body of main is generated twice?

ghaith, 2022-05-31:
> this looks wrong to me (in main)
> 
> `%4` is an i8* to an allocated parameter-char (81xi16) which will be passed to the function `%5` is an i8* to the literal "hello" (6xi16)
> 
> then we copy 80 bytes from %5 into %4. - 80 bytes looks wrong to me. should it copy 12 (=2*6) bytes?
> 
> ```llvm
>   %4 = bitcast [81 x i16]* %1 to i8*
>   %5 = bitcast [6 x i16]* %3 to i8*
>   call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %4, i8* align 1 %5, i32 80, i1 false)
> ```
> 
> and it looks like the body of main is generated twice?

The main is generated once, but yes  I noticed the copying of 80.
I'll attach code to the issue later

ghaith, 2022-06-01:
So the main issue seems to be the interpolation with Rust
The rust function expects a wrapper to the u8 array as we cannot send raw arrays by value using ffi.
I suggest we start working on support for string pointers for functions. This will be a general issue for all arrays passed to external functions, so the STRING(*) and ARRAY[*] types would make sense here, where the values are always pointers.
I would keep the normal STRING in functions by values as there might be existing code that relies on knowing the string size.

closed 2022-06-10 by ghaith