PLC-lang/rusty #509: Generic function return type not working

99NIMI, 2022-06-22:
It's not working when the generic type is only specified for the return type of a function

riederm, 2022-06-22:
modern languages use type-inference for this, so they look what you're assigning this to to guess the proper return type

e.g. in rust:
```rust
let x : u32 = foo(); // works, so foo will return u32 because x's type is clear
```
```rust 
let x = foo(); // illegal! because x's type is not clear
```

@99NIMI do we need this for a standard-function or something?

99NIMI, 2022-06-22:
> modern languages use type-inference for this, so they look what you're assigning this to to guess the proper return type
>
> e.g. in rust:
>
> ```rust
> let x : u32 = foo(); // works, so foo will return u32 [...]

@riederm there are some standard-functions with e.g ANY_INT return type
the first use case was for [DAY_OF_WEEK](https://normen.bachmann.at//DIN-EN-ISO/DIN_EN_61131_3.pdf#page=108&zoom=100,0,0)
currently we will always return SINT because the result will always be 0-6
there are other functions with a bigger range of the possible result, for that functions we could return the largest needed type but i don't think that this would be a clean approach to solve this

closed 2022-08-30 by ghaith