PLC-lang/rusty #512: Generics called "T" confuse the resolver/indexer

ghaith, 2022-07-06:
**Describe the bug**
Since `T` is an alias for `TIME`, our generics code ends up resolving T to time instead of the generic
I noticed this internally for builtin functions where the cast_if_needed was always casting DINTs to i64 (TIME)

riederm, 2022-07-08:
I think we could even talk about removing the T-alias for TIME since ...

a) it is totally arbitary
b) TIME is not an awfully long name
c) it does not make the code more readable
d) there are only 1-letter aliases for the time-related datatypes
e) for compatibility projects you can easily add the alias-type declaration yourself

ghaith, 2022-07-11:
I think this is solved in the currently open PR.
The builtins were not being run through the preprocessor before indexing, the type T should have been something like foo_T for a generic.

closed 2022-08-30 by ghaith