PLC-lang/rusty #514: Generic CHAR functions are not easy to derive

ghaith, 2022-07-11:
**Describe the bug**
If we are working with a generic function being called with a `CHAR`, the type cannot easily be derived
Example : 
```st
FUNCTION limit<T : ANY> : T
VAR_INPUT
   min,in,max : T;
END_VAR
END_FUNCTION
FUNCTION main : DINT
VAR
  b : CHAR;
END_VAR
   limit('a',b,'c');
END_FUNCTION
```
'a' and 'c' will not be resolved as char, but probably as string, to correctly call this function we need something like 
```st
limit(CHAR#'a', b, CHAR#'c');
```

We should consider other options

riederm, 2022-07-11:
other languages use the different quotes to dustinguish string & char (e.g. c, java, etc.). Since the ST-standard does not distinguish between char and string-literals I think the CHAR#'a' is a valid thing.

so 'a' and "a" should both be a strings  (utf8 & utf16) in my opinion.

ghaith, 2022-07-13:
you're right, it's very hard to guess this needed chars at the moment, I just thought it will make code harder to read for these kind of functions.
I'll close it.

closed 2022-07-13 by ghaith