PLC-lang/rusty #551: run [cargo bulid --release]------LLVM binaries specified by LLVM_SYS_130_PREFIX are the wrong version.

hz0908, 2022-08-24:
**Describe the bug**
My system is ubuntu 20. 04-4. Today I followed the methord without useing a docker file to bulid and install rusty . But after I install all 
packages(include-sudo apt-get install -y llvm-12-dev) the project needs, it went wrong when I use the command--[cargo bulid --release]. The terminal told me that [LLVM binaries specified by LLVM_SYS_130_PREFIX are the wrong version.] I dis use [export LLVM_SYS_130_PREFIX=/usr/lib/llvm-12] to ponit the LLVM. 

**To Reproduce**
sudo apt-get install -y llvm-12-dev
export LLVM_SYS_130_PREFIX=/usr/lib/llvm-12
cargo bulid --release
wrong:LLVM binaries specified by LLVM_SYS_130_PREFIX are the wrong version.

**Expected behavior**
Yesterday, someone told me that the documentation I fllowed is outdate, so i guess that now the project needs llvm-13 ?

**Additional context**
nothing but thanks

**Tests**
Optional: Add a link to a PR or a branch where you have a failing test of the issue.

ghaith, 2022-08-25:
We do indeed require llvm 13 to build now.
For reference you could use the docker file inside the dev container to find the packages that need to be installed : 
https://github.com/PLC-lang/rusty/blob/master/.devcontainer/Dockerfile
In general, the packages are llvm, clang, lld in addition to liblld and libclang.

I will keep this issue as a tracker for updating that part of the documentation.

hz0908, 2022-08-25:
Thank you very much 

黑蛇
***@***.***

 

 https://github.com/PLC-lang/rusty/blob/master/.devcontainer/Dockerfile

closed 2022-08-26 by ghaith