PLC-lang/rusty #556: EXPT wrong type

99NIMI, 2022-08-29:
````
FUNCTION EXPT<U : ANY_NUM, V: ANY_NUM> : U
VAR_INPUT
    ELEMENT: U;
    EXPONENT: V;
END_VAR
END_FUNCTION
```
Currently EXPT will return a real type when the EXPONENT is not a LinteralInteger

ghaith, 2022-08-29:
Do you mean if it's a variable declared as INT as well? Because for other types this is by design

99NIMI, 2022-08-29:
Yes, because at one point we check if both ELEMENT and EXPONENT are DataTypeInformation::Integer
and if the EXPONANT is SIGNED we check extra if its positive, but on this check we expect to have an AstStatement::LiteralInteger alse we return false
So when we use a reference as EXPONENT and its SIGNED-INT we won't pass this check and then we try to get a real-type

ghaith, 2022-08-29:
I would say this is by design since we can't find the value at compile time, and negative exponents will always have a REAL value 
$2^{-2} = \frac{1}{2^2} = 0.5$
Maybe we should add tests/doc explaining that

99NIMI, 2022-08-29:
Yeah i faced that issue with not enough info at compile time, but there are i think some in the moment ignored tests that state the same problem
but i can look in to them and fix the tests/doc and close this issue again

ghaith, 2022-08-29:
Yes, I think this would be the better solution. The other option (more consistent) is to make all expt functions return floats, but I would like to keep the INT/Positive INT functions returning INTs here.
So we could add an explanation in the tests of why the DINT variable is actually going to result in a REAL (since it could be negative) We should also make sure UDINT and other unsigned values are returning INT in that case.

mhasel, 2022-12-15:
This also happens for type casts, which are known at compile time.

``2**ULINT#23;`` will result in a REAL value.

Related issue: #681

99NIMI, 2023-01-26:
EXPT moved to StandardFunctions