PLC-lang/rusty #561: Building rusty without docker Failed on Ubuntu 20.0.4

hz0908, 2022-08-30:
**Describe the bug**
When I followed the new documentation to bulid rusty, it failed and showed that:

warning: wrapper/lld-c.cpp:2:10: fatal error: lld/Common/Driver.h: There is no such file or directory
warning:     2 | #include <lld/Common/Driver.h>
warning:       |          ^~~~~~~~~~~~~~~~~~~~~
warning: compilation terminated.

error: failed to run custom build command for `lld_rs v130.0.0 (https://github.com/mun-lang/lld-rs?branch=main#acc3ec6b)`

**To Reproduce**
Steps to reproduce the behavior:
_If you are reporting a compile issue, it would be good to include some code here_
curl https://sh.rustup.rs -sSf | sh                                        #install rust using a shell
sudo apt-get install build-essential                                   #the usual standard build tools
#########
bash -c "$(wget -O - https://apt.llvm.org/llvm.sh)"
wget https://apt.llvm.org/llvm.sh
chmod +x llvm.sh
sudo ./llvm.sh 13                                                              #install llvm13   
#########
export LLVM_SYS_130_PREFIX=/usr/lib/llvm-13              #set env
sudo apt-get install -y libz-dev                                       #zlib (apt: libz-dev)
sudo apt-get install -y libclang-common-13-dev           #Polly in the form of a static library (included in apt package libclang....
cd ./rusty2/rusty                                                              #rusty2 is a folder I found 
sudo apt  install cargo                                                    #install cargo
cargo build --release  

then the warning and erro accured

**Expected behavior**
A clear and concise description of what you expected to happen.
bulid successfully

**Additional context**
Add any other context about the problem here.
I think it may not be a good question, because it is too specific. But I am a beginnner of linux and rusty, I dont konw how to make it. So i ask here, if you think there is no need to answer, just tell me. i totally understand. Thank you !
**Tests**
Optional: Add a link to a PR or a branch where you have a failing test of the issue.

ghaith, 2022-08-30:
Hello, questions are always welcome :) 
If you have tried building the same repo before installing llvm, try doing a cargo clean first.

hz0908, 2022-08-31:
> 
Thanks. But after I did a cargo clean, I tried [cargo build --release ]. It also failed with the same wrong---[failed to run custom build command for lld_rs v130.0.0 (https://github.com/mun-lang/lld-rs?branch=main#acc3ec6b)]

ghaith, 2022-08-31:
I will try out the build a new VM to see If we are missing a step. You now have a clang-13 command I assume? and the liblld and libclang packages were installed correctly?
I recently installed on ubuntu and it worked as expected but as I'm used to the steps it could have been that I did something automatically without noticing.

Also if you want an easier way to checkout the source and compile it, you can use the dev container, inside visual studio code, to do this on ubuntu install vscode, and the remote container extension, then install docker on ubuntu (`sudo apt install docker.io`) and add your user to the docker group  : `sudo usermod -aG docker ${USER}`
When you then open the rusty workspace in vscode, it will offer to open it in a dev container.

if you just want the binaries, you can also find them here : https://github.com/PLC-lang/rusty/actions/workflows/rust.yml?query=branch%3Amaster under the latest build, at the bottom there would be a windows_debug and linux_debug of the latest master branch

hz0908, 2022-09-01:
> I will try out the build a new VM to see If we are missing a step. You now have a clang-13 command I assume? and the liblld and libclang packages were installed correctly? I recently installed on ubuntu and it worked as expected but as I'm used to the steps it could have been that I did something automatically without noticing.
> 
> Also if you want an easier way to checkout the source and compile it, you can use the dev container, inside visual studio code, to do this on ubuntu install vscode, and the remote container extension, then install docker on ubuntu (`sudo apt install docker.io`) and add your user to the docker group : `sudo usermod -aG docker ${USER}` When you then open the rusty workspace in vscode, it will offer to open it in a dev container.
> 
> if you just want the binaries, you can also find them here : https://github.com/PLC-lang/rusty/actions/workflows/rust.yml?query=branch%3Amaster under the latest build, at the bottom there would be a windows_debug and linux_debug of the latest master branch
> 
> You could also get going relatively quickly by visiting : https://gitpod.io/#https://github.com/PLC-lang/rusty/ This will open the development environment in the cloud using the dev container (docker)
Thanks，when it comes to clang-13、liblld and libclang package，I can‘t find the specific information about them and I think maybe they are all included in llvm13?  And I install llvm13 use the auto-install shell from the website-[https://apt.llvm.org/]-  you recommended in #557 
the commands I use are as follows:
**bash -c "$(wget -O - https://apt.llvm.org/llvm.sh)"
wget https://apt.llvm.org/llvm.sh
chmod +x llvm.sh
sudo ./llvm.sh 13 #install llvm13**

But I don't konw if I  have  clang-13, liblld and libclang packages. 

And I have tried the method using a docker file to insatall rusty. It works very well, but now I just want to try another method.

Thanks for your timely help .

hz0908, 2022-09-01:
Thanks，when it comes to clang-13、liblld and libclang package，I can‘t find the specific information about them and I think maybe they are all included in llvm13? And I install llvm13 use the auto-install shell from the website-[https://apt.llvm.org/]- you recommended in https://github.com/PLC-lang/rusty/issues/557
the commands I use are as follows:
bash -c "$(wget -O - https://apt.llvm.org/llvm.sh)"
wget https://apt.llvm.org/llvm.sh
chmod +x llvm.sh
sudo ./llvm.sh 13 #install llvm13

But I don't konw if I have clang-13, liblld and libclang packages.

And I have tried the method using a docker file to insatall rusty. It works very well, but now I just want to try another method.

Thanks for your timely help .

ghaith, 2022-09-01:
This will only get you the llvm-13 and probably clang, but not the extra dev dependencies, you would still need to install the extra dependencies mentioned [here](https://plc-lang.github.io/rusty/build_and_install.html) manually
These would be `llvm-13-dev`, `liblld-13-dev`, `libz-dev`, `libclang-common-13-dev`
After that you should be able to do a `cargo clean` and then build successfully

closed 2022-12-07 by ghaith