PLC-lang/rusty #582: Override cannot be used as field name

ghaith, 2022-09-22:
This is a spinoff of #249 
I saw some existing code that uses Override as a field name, I believe Oscat also has something similar.

We need a decision on how to support these, either request existing projects to change those keywords, or support a way for keywords to also be identifiers.

riederm, 2022-09-22:
i'd like to discuss the options that we have to allow keywords as identifiers and the implications:
so how can we teach the parser to allow keywords as identifiers - I think there are two options:

1) there are no keywords:
Remove keywords from the lexer and parse an identifier wherever we expect a keyword and do string-comparison to identify keywords.

2) where-ever we allow an identifier in the parser, we must also allow (most) keywords: 
like `if (lexer.current_token == identifier || is_keyword(lexer.current_token) {`

volsa, 2026-02-27:
Some duplicate issue exists for this, closing this one

closed 2026-02-27 by volsa