PLC-lang/rusty #602: Redefining constant variables does not cause an error

mhasel, 2022-10-12:
As of now, the following code compiles and validates without error. I'm not sure this is how it should behave. In my opinion, it should not be possible to redefine constants (at the very least not in the same scope they are defined in). Please discuss.

```
        FUNCTION add : DINT
        VAR CONSTANT
            TWO : DINT := 2;
            THREE : DINT := 3;
        END_VAR
        VAR
            TWO : DINT := 4;
            THREE : DINT := 5;
        END_VAR
            add := TWO + THREE;
        END_FUNCTION
```
```
          7 │+define i32 @add() {
          8 │+entry:
          9 │+  %TWO = alloca i32, align 4
         10 │+  %THREE = alloca i32, align 4
         11 │+  %add = alloca i32, align 4
         12 │+  store i32 4, i32* %TWO, align 4
         13 │+  store i32 5, i32* %THREE, align 4
         14 │+  store i32 0, i32* %add, align 4
         15 │+  %load_TWO = load i32, i32* %TWO, align 4
         16 │+  %load_THREE = load i32, i32* %THREE, align 4
         17 │+  %tmpVar = add i32 %load_TWO, %load_THREE
         18 │+  store i32 %tmpVar, i32* %add, align 4
         19 │+  %add_ret = load i32, i32* %add, align 4
         20 │+  ret i32 %add_ret
         21 │+}
```

riederm, 2022-10-12:
yes this is a problem. qualified names must be unique for the whole application.
Qualified names here are: "add.TWO" and "add.THREE".

This should be solved by allowing the index to store more than one entry per name (use multi-maps instead of hashmap). Every multi-entry is a problem.

With the current implementation the first "add.TWO" will be overwritten in the index by the next "add.TWO"

riederm, 2022-10-12:
related issues: 
#354 (duplicate functions should be a problem)
https://github.com/PLC-lang/rusty/issues/285 (location information in the index) -> required to report meaningful problems

mhasel, 2022-10-13:
possibly also related, depending on the approach we want to take with enums.

[#569](https://github.com/PLC-lang/rusty/issues/569)  (Global variable initializer type does not match global variable type)

ghaith, 2022-10-17:
> possibly also related, depending on how we want to approach multiple enumerations of the same name.
> 
> [#569](https://github.com/PLC-lang/rusty/issues/569) (Global variable initializer type does not match global variable type)

I think #569 is not related, in there we need to support multiple enum variants with the same name if they are defined in different enums, the enums can then be accessed using Enum#Variant

PR #622 (2022-11-04): 354 duplicate functions should be a problem
Introduced SymbolMap which is a MultiMap that keeps the insertion order alive when iterating the elements. That way we keep exisiting tests working (I tried the multimap-create and it mixes up the order of stuff in the index which results in different order of elements generated in IR, so basically 80% of all tests would require a review).

Furthermore I introduced the global_validator.rs which validates stuff using the index. This commit extends Diagnostics to more than one location. first location remains the diagnostics main location, following lo [... cut]
closed 2022-11-04 by riederm