PLC-lang/rusty #634: unresolved constant variable

rarris, 2022-11-07:
**program example**
```
FUNCTION main : DINT
    VAR CONSTANT
    	cmd1 : command;
        var2 : INT;
    END_VAR

END_FUNCTION

TYPE commands :
STRUCT
	ReInit : BOOL;
	Reset : BOOL;
	Stop : BOOL;
END_STRUCT
END_TYPE
```

**error**: Unresolved constant 'cmd1' variable
   ┌─ test.st:19:6
   │
19 │       cmd1 : command;
   │       ^^^^^

**error**: Unresolved constant 'var2' variable
   ┌─ test.st:20:6
   │
20 │       var2 : int;
   │       ^^^^^

riederm, 2022-11-07:
cmd1 & var2 need an initial value - the compiler message is really bad. We should improve the error message for this situation. Or do you think this code should work like this? the constants never get a value - or should they get default values?

riederm, 2022-11-07:
by the way - there is a typo I fixed so be careful when you test this again: I changed `cmd1: command` to `cmd1: commands` (adding the s in the end). 

so the code would work like this:
```st
FUNCTION main : DINT
    VAR CONSTANT
    	cmd1 : command;
        var2 : INT;
    END_VAR

END_FUNCTION

TYPE commands :
STRUCT
	ReInit : BOOL;
	Reset : BOOL;
	Stop : BOOL;
END_STRUCT
END_TYPE
```

ghaith, 2022-11-08:
Uninitialized constants should also get the default value. I've seen code that defines an uninitialized `STRUCT` as a "Reset" constant to easily reset other values during execution.

ghaith, 2022-11-28:
This is still not solved, the code in the example still fails on codegen (Struct init does not get an initial value)

commit c437ca99 (2022-11-12): uninitialized consts fall back to default value

constants with no initializers used to cause validation errors, from
this commit on they get thei'r datatypes default value.

note that alias and range-types will inherit their default from their
referenced types.

fixes #634
PR #645 (2022-11-24): uninitialized consts fall back to default value
constants with no initializers used to cause validation errors, from this commit on they get thei'r datatypes default value.

note that alias and range-types will inherit their default from their referenced types.

fixes #634
commit 3dc10fbf (2022-11-24): uninitialized consts fall back to default value (#645)

* uninitialized consts fall back to default value

constants with no initializers used to cause validation errors, from
this commit on they get their datatypes default value.

note that alias and range-types will inherit their default from their
referenced types.

added several test-szenarios
* test that shows the transitivity for range types
* test that covers enum default const evaluationF

fixes #634
commit 91498fc6 (2022-11-30): initialize local VAR CONSTANT function vars

aggregate local constant variables (in functions) used to have
problems if they were not properly initialized. Instead of forcing an
initializer, we fallback to the type's default value or zeroinitializer.

added regression test for a previously failing situation

fixes #634
closed 2022-12-02 by riederm