PLC-lang/rusty #636: lld: error: undefined symbol: memcpy

mhasel, 2022-11-08:
**Describe the bug**
when trying to assign a VAR_INPUT string to a VAR_IN_OUT buffer string, compilation fails with 
`Error: GeneralError { message: "lld: error: undefined symbol: memcpy\n>>> referenced by main\n>>>               /tmp/.tmpQfffq6/prog:(copy)\n", err_no: linker__generic_error }`

**To Reproduce**
```
FUNCTION main : DINT
  VAR
    s1: STRING[2048] := 'test';
    s2: STRING[2048];
  END_VAR
    copy(s1, s2);
END_FUNCTION
FUNCTION copy : DINT
  VAR_INPUT {ref}
    in: STRING[2048];
  END_VAR
  VAR_IN_OUT
    out: STRING[2048];
  END_VAR
    out := in;  
END_FUNCTION
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Tests**
Optional: Add a link to a PR or a branch where you have a failing test of the issue.

ghaith, 2022-11-08:
When compiling an executable, the embedded ldd needs more information that we give by default, for example an entry point, this means linking a `crt0` file, and additionally specifying the location of the `memcpy` (the system's libc)  
The easiest way to avoid this for now it to use a C compiler as a linker, both `gcc` and `clang` are compiler drivers and will trigger linking after the "compilation" is done. You can use the `--linker=cc` or `--linker=clang` flag for that

mhasel, 2022-11-08:
This happens even when specifying a linker. Here is the build-and-run-script I am using:
```
#! /bin/bash
./rustyc build --linker=cc
export LD_LIBRARY_PATH="./lib"
./build/prog
```
I can try it with clang instead and see if that changes things

mhasel, 2022-11-08:
I have tried it again and it worked first try.

closed 2022-11-08 by mhasel