PLC-lang/rusty #652: Performance problems with string functions

ghaith, 2022-11-14:
**Is your refactor request related to a problem? Please describe.**
When declaring a lot of STRING functions in a project, especially with big sizes, we need to reserve data for all these functions, resulting in extremely slow compile times. We noticed this in the standard functions where we defaulted for a [2048] string length, we reverted now to 80 there, this is however not practical because strings of bigger sizes will always be truncated.

**Describe the solution you'd like**
We need to find a better way to handle strings, one idea is to use implicit output variables whenever we see a string, and delegate the allocation to the caller, hence using a stack allocation instead of reserving global memory. We tried similar solutions before but they were problematic. we might want to give it another shot though.

We can also alternatively treat all strings as pointers to a memory that will be allocated later, but we need to define when later is.
Another idea is to use a ring buffer to handle strings but this is very error-prone and I think belongs more in a runtime rather than a compiler.

ghaith, 2022-11-15:
Upon investigation,  it seems that using the VAR_OUTPUT solution would work, it generates much smaller binaries and compiles much faster.

riederm, 2022-11-16:
We're going to avoid aggregate return-types. This means that we need to change the way we return strings, structs and arrays from functions.

implementation strategy for declarations:
- this implementation only considers functions with aggregate return types (strings, structs and arrays)
- the return value is generated as an out-parameter
- the function now returns VOID
e.g.
```st
FUNCTION CONCAT : STRING[1024]
   ...
END_FUNCTION
```

implementation strategy for calls to such functions:
- the target of the call (assignment-target) will be passed as the out-parameter
`result := concat(a,b);` --> `concat(&result, a, b);`
- we should store the result in a temporary variable if the target is an rvalue:
`CONCAT(LEFT(a, 10), RIGHT(a, 10));` -> so the results of left and right need to be stored in temporary variables and therefore allocated on the stack of the caller.

commit e81b8871 (2022-11-24): use by-ref out variable for aggreate return types

Functions that return an aggregate type (string, array, struct) no
longer return the value as the function's return type, they rather
return the value via an out-pointer that needs to be passed by the
caller.
old: `myArr := foo(1, 2, 3);`
will be generated automatically as: `foo(&myArr, 1, 2, 3);`

The out parameter is inserted at position 0 into the parameter list.

To properly handle these situations this commit introduces the
ExpressionValue in expression_generator.rs to easily distinguish
lvalues (something we can get a pointer to) from rv [... cut]
commit 4960d5e3 (2022-11-24): use by-ref out variable for aggreate return types

Functions that return an aggregate type (string, array, struct) no
longer return the value as the function's return type, they rather
return the value via an out-pointer that needs to be passed by the
caller.
old: `myArr := foo(1, 2, 3);`
will be generated automatically as: `foo(&myArr, 1, 2, 3);`

The out parameter is inserted at position 0 into the parameter list.

To properly handle these situations this commit introduces the
ExpressionValue in expression_generator.rs to easily distinguish
lvalues (something we can get a pointer to) from rv [... cut]
commit bb5e035f (2022-11-24): use by-ref out variable for aggreate return types

Functions that return an aggregate type (string, array, struct) no
longer return the value as the function's return type, they rather
return the value via an out-pointer that needs to be passed by the
caller.
old: `myArr := foo(1, 2, 3);`
will be generated automatically as: `foo(&myArr, 1, 2, 3);`

The out parameter is inserted at position 0 into the parameter list.

To properly handle these situations this commit introduces the
ExpressionValue in expression_generator.rs to easily distinguish
lvalues (something we can get a pointer to) from rv [... cut]
commit e57b73fe (2022-11-25): use by-ref out variable for aggreate return types

Functions that return an aggregate type (string, array, struct) no
longer return the value as the function's return type, they rather
return the value via an out-pointer that needs to be passed by the
caller.
old: `myArr := foo(1, 2, 3);`
will be generated automatically as: `foo(&myArr, 1, 2, 3);`

The out parameter is inserted at position 0 into the parameter list.

To properly handle these situations this commit introduces the
ExpressionValue in expression_generator.rs to easily distinguish
lvalues (something we can get a pointer to) from rv [... cut]
closed 2022-12-02 by ghaith