PLC-lang/rusty #659: array of struct initialization problem

rarris, 2022-11-21:
```
PROGRAM mainProg
VAR
END_VAR

VAR CONSTANT
offset_check : INT := 1;
slots_nr : INT := 7;

config : ARRAY[offset_check..slots_nr] OF INT :=
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5),
(var1:=1, var2:=2, var3:=5, channel:=10);
END_VAR
;
END_PROGRAM
```

```
TYPE struct1 :
STRUCT
	var1				: DINT;
	var2			: DINT;
	var3				: DINT;
	channel		: ARRAY[1..16] OF DINT; 
END_STRUCT
END_TYPE
```

error: Cannot generate literal initializer for 'mainProg.config': Value cannot be derived
Error: SyntaxError { message: "Some initial values were not generated", range: [SourceRange { range: 0..0 }], err_no: codegen__general }

rarris, 2022-11-30:
the error "Some initial values were not generated" was fixed, but now we have another error

```
Error: SyntaxError { message: "no type hint available for Assignment {\n
left: Reference {\n   name: \"var1\",\n    },\n
right: LiteralInteger {\n        value: 1,\n    },\n}"
```

same code example

99NIMI, 2022-11-30:
> the error "Some initial values were not generated" was fixed, but now we have another error
> 
> ```
> Error: SyntaxError { message: "no type hint available for Assignment {\n
> left: Reference {\n   name: \"var1\",\n    },\n
> right: LiteralInteger {\n        value: 1,\n    },\n}"
> ```
> 
> same code example

there was one little issue in the code example above, the ARRAY was defined with type of INT, should be struct1
now the code should be correct
i didn't get any issue trying to compile this example 
the IR also seems correct to me

rarris, 2022-12-01:
yes true ... my bad :)
but the issue is still reproducible - i've updated the code in the first post
if we write instead of _global_const_ anything else, a dummy non declared var or so, we got the same error, no validation there

closed 2022-12-12 by 99NIMI