PLC-lang/rusty #685: Change styleguide

volsa, 2022-12-07:
@ghaith and I discussed the possibility of introducing a [`rustfmt.toml`](https://rust-lang.github.io/rustfmt/?version=v1.5.1&search=) file to change RuSTys style guidelines and make the code more readable. Specifically we talked about 
1. increasing the [`max_width`](https://rust-lang.github.io/rustfmt/?version=v1.5.1&search=#max_width) value to something like `110` or
2. using [`use_small_heuristics = Max`](https://rust-lang.github.io/rustfmt/?version=v1.5.1&search=#use_small_heuristics) in combination with 1. or
3. using 1. in combination with some additional options specified in `use_small_heuristics` (see the first [paragraph](https://rust-lang.github.io/rustfmt/?version=v1.5.1&search=#use_small_heuristics))

riederm, 2022-12-07:
are there some reasonable options that other big rust-projects established? Or how did you pick the suggested options?

volsa, 2022-12-07:
Kind of, many bigger projects[1] tend to use `use_small_heuristics = "Max"` only because the default value used by rustfmt, i.e. `100`, is generous enough. I like to use `110` in my personal projects because Rust code can get somewhat lengthy and one character alone can ruin formatting, which a lot of people seem to [agree on](https://www.reddit.com/r/rust/comments/xreg1a/comment/iqegsfa/?utm_source=share&utm_medium=web2x&context=3). The issue why Rust code tends to get ugly fast, however, is because of the default [ratios](https://rust-lang.github.io/rustfmt/?version=v1.5.1&search=#use_small_heuristics) established by rustfmt which is eliminated with the `Max` value. That being said `use_small_heuristics` isn't all sunshine and rainbows depending on who you ask, e.g. take the following code snippets:

- `use_small_heuristics = "Default"` 
  ```Rust
      let struct_foo = Foo {
          member_one: 1,
          member_two: 2,
      };
  
      let struct_bar = Bar {
          member_one: 1,
          member_two: 2,
          member_three: 3,
          member_four: 4,
          member_five: 5,
      };
  ```
- `use_small_heuristics = "Max"` 
```Rust
    let struct_foo = Foo { member_one: 1, member_two: 2 };

    let struct_bar =
        Bar { member_one: 1, member_two: 2, member_three: 3, member_four: 4, member_five: 5 };
```
In my opinion `struct_foo` is more pleasant to read in the lattter and `struct_bar` is in the former. That's why I also suggested option 3 _if_ we would want to fine-tune this but I assume big projects don't do this for consistency stake?

[1]: 
- [Rust (internal crates)](https://github.com/search?q=repo%3Arust-lang%2Frust+use_small_heuristics+language%3ATOML&type=code&l=TOML)
- [Rust Analyzer](https://github.com/rust-lang/rust-analyzer/blob/master/rustfmt.toml)
- [Alacritty](https://github.com/alacritty/alacritty/blob/master/rustfmt.toml#L8)
- [MeiliSearch](https://github.com/meilisearch/meilisearch/blob/main/.rustfmt.toml#L3)

volsa, 2022-12-13:
We've settled with option 2, i.e.
```toml
max_width = 110
use_small_heuristics = "Max"
```

closed 2022-12-16 by ghaith