PLC-lang/rusty #700: Move the examples into own repository

ghaith, 2022-12-16:
The current examples folder is a mess, i would suggest deleting it and starting with new specific examples in a new repo
These examples can then serve as showcases and help.

teksturi, 2023-05-05:
I would say that if examples are in own repo no one will ever look those. I would say that no one will even edit those. Editable code needs to be close to those who edit them. In my opinion it would be good if examples would inside documentation. This way it is easy to add text and explanations `Note: This syntax is difference than TwinCat`. Documentation also have search functionality out of the box.

volsa, 2023-05-08:
Hi, thanks for your input! 
We've internally discussed this some while ago and mostly came to the same conclusion as you. That is we want to keep the same structure we have now, with the exception of adding README(s) inside the `examples/` folder to give further context. As a result the structure might either end up like
```
examples/
  foo.st
  bar.st
  ...
  README.md
```
or 
```
examples/
  foo/foo.st
  foo/README.md
  bar/bar.st
  bar/README.md
  ...
```

cc @ghaith