PLC-lang/rusty #706: Assigning VAR_INPUT variable should not be possible

99NIMI, 2022-12-22:


r2k-in-the-vortex, 2022-12-28:
Standard(2003) explicitly states that VAR_INPUT is "Externally supplied, not modifiable within organization unit". 

But TwinCAT/Codesys at least has no complaints assigning VAR_INPUT from any scope, MATIEC will happily do it too, I think many PLC-s don't really care about that rule as it's rather poorly thought out anyway.

Even if you can't do `invar := something;` you can do `ADR(invar)^:= something;` or `THIS^.invar := something;` for the same effect. Maybe downgrade that error to style warning or something?

99NIMI, 2022-12-29:
I updated the title to VAR_INPUT {ref}
currently it's possible and assigning a VAR_INPUT {ref} variable would behave like VAR_IN_OUT
```
FUNCTION main : DINT
VAR
    x : INT := 10;
END_VAR
    prog(x);
    // x is now 20
END_FUNCTION

PROGRAM prog
VAR_INPUT {ref}
    in1 : INT;
END_VAR
    in1 := 20;
END_PROGRAM
```
and i don't think that we should allow this behavior, that's the use-case for VAR_IN_OUT

for normal VAR_INPUT why not, could be useful in some cases, you can do this in other languages too
but a warning would be also fine for me

closed 2023-05-24 by volsa