PLC-lang/rusty #707: Cannot generate literal initializer for 'mainProg.config': Value cannot be derived

rarris, 2022-12-22:
```
PROGRAM mainProg
VAR
END_VAR
VAR CONSTANT
config : ARRAY[1..2] OF struct1 :=
(var1:=1),
(var1:=1, channel:=10(global_const), 8(global_const));
END_VAR
END_PROGRAM
```

```
TYPE struct1 :
STRUCT
   var1 : DINT;
   channel : ARRAY[1..16] OF DINT; 
END_STRUCT
END_TYPE
```

```
VAR_GLOBAL CONSTANT
global_const : DWORD  := 16#0001;
END_VAR
```

error: Cannot generate literal initializer for 'mainProg.config': Value cannot be derived
Error: SyntaxError { message: "Some initial values were not generated", range: [SourceRange { range: 0..0 }], err_no: codegen__general }

99NIMI, 2022-12-29:
@ghaith @riederm 
as rares described the array initialization is the problem
1: `channel:=10(global_const), 8(global_const)` -> doesn't work
2: `channel:=(10(global_const), 8(global_const))`-> works after we added the ()
3: `arr : ARRAY [1..2] OF DINT := 1, 2, 3;`-> works without the ()
the problem is how our parser works

why option 3 works, if our parser matches the ARRAY keyword, no matter what comes after `:=` we will generate an ExpressionList for it, until we match for semicolon -> this is the reason we don't need to use () for the initializer

why option 1 does't work!, there is no ARRAY keyword in this context, we would only parse for an ExpressionList if we match `(`

the question is how should we handle this ? should we modify the parser and also generate an ExpressionList for something like this -> `1,2,3;` until we match semicolon, this would fix the error
or
is this an valid error and () is needed in this case

riederm, 2022-12-29:
yes thats the parsers fault ...
- `channel:=10(global_const), 8(global_const)`
the parser eagerly eats the assignment on the left side and creates an expression list like this:
```
ExpressionList [ 
   Assignment(
      Reference, 
      RepeatExpression),  
   RepeatExpression
]
```
while it should create an AST like this:
```
Assignment(
   Reference, 
   ExpressionList [
      MultipliedStatement, 
      MultipliedStatement]
)
```

riederm, 2022-12-29:
so in short, my vote is:
`this is a valid error and () is needed in this case`. Without brackets, the coma starts the assignment to the next struct field.

what do you guys think?

99NIMI, 2022-12-29:
my first thought was also that this is an error
array initialization without `()` feels a little bit unnatural to me, normally you would see something like `{1,2,3,...}`in other languages
and i don't think the effort would be worth it

99NIMI, 2023-01-03:
```
error: Array initializer must be an initializer list!
   ┌─ main.st:29:2
   │
29 │   arr := 1, 2;
   │   ^^^ Array initializer must be an initializer list!

error: Expected identifier or '('
   ┌─ main.st:29:12
   │
29 │   arr := 1, 2;
   │             ^ Expected identifier or '('
```

current implementation of the validatio will report the error like C

rarris, 2023-03-24:
apparently we have another type of struct initialization ...

```
PROGRAM mainProg
VAR
var_init : ARRAY[0..15] OF STRUCT1 := (name:='text'),(node:=1),(myArr:=(x1:=FALSE,x2:=TRUE));
END_VAR
END_PROGRAM
```

```
TYPE STRUCT1 :
	 STRUCT
		var10 : BOOL;
		name : STRING[31];
		node : DINT;
		number : DINT;
		myArr : ARRAY[0..10] OF STRUCT2;
	END_STRUCT
END_TYPE
```

```
TYPE STRUCT2 :
	 STRUCT
		x1 : BOOL;
		x2 : BOOL;
		x3 : DINT;
		x4 : DINT;
	END_STRUCT
END_TYPE
```

error: Cannot generate literal initializer for 'mainProg.var_init': Value cannot be derived
Error: SyntaxError { message: "Some initial values were not generated", range: [SourceRange { range: 0..0 }], err_no: codegen__general }

closed 2023-09-05 by volsa