PLC-lang/rusty #710: action call without "()"

rarris, 2022-12-23:
```
PROGRAM mainProg
VAR
END_VAR
MyAction;
END_PROGRAM
ACTIONS
ACTION MyAction
END_ACTION
END_ACTIONS
```

Error: SyntaxError { message: "Could not resolve reference to MyAction"

should we accept action calls without parentheses?

riederm, 2022-12-29:
I hate that codesys does this, so I guess we need to support it.
It's not that hard. The main issue for the code-generator is, that as far as the parsers knows, this is a Reference, not a call-statement.
- the Resolver should have annotated the parsed reference `MyAction` with a proper statement annotation that already points to the Action. It should have annotated it with:
```
StatementAnnotation::Program { 
   qname: "mainProg.MyAction" 
}
```
while a "normal" Reference to a variable would have been annotated with a `StatementAnnotation::Variable`.

- the expression_generator now needs to look for references, that are annotated with `StatementAnnotation::Program` or `StatementAnnotation::Function` - which is also allowed i guess.
we should look for `StatementAnnotation::Program` or `StatementAnnotation::Function` here
https://github.com/PLC-lang/rusty/blob/25d7a9edf6f591a69dd8e3381053b4c33585da51/src/codegen/generators/expression_generator.rs#L208-L218

- while it is a smell in the first place, the parser strongly distinguishes between a Reference `x` and a qualified Reference `x.y`, this now becomes an advantage for this issue. We don't need to worry that a qualified reference to a program's member (`MainProg.member`) will suddenly be handled as a call.

- we already have some helper code to fake a call-statement where there is none. see
https://github.com/PLC-lang/rusty/blob/25d7a9edf6f591a69dd8e3381053b4c33585da51/src/ast.rs#L1270-L1277
you can just create a call-statement on the fly, we know the call-name from the reference's annotation, reuse the reference's ID, parameters are empty, (so also no id is required - use 0), location is the one from the reference. And here you go, you have a brand new call statement that can be sent to codegen.

ghaith, 2022-12-29:
Do we really _have_ to really support it? I see it more as a bug in codesys
than a feature.

riederm, 2022-12-30:
> Do we really have to really support it? I see it more as a bug in codesys than a feature.

I understand your point, I would definitely go with a warning-validation if we end up supporting it. We went through the hassle when we did the PLC-Developer.

volsa, 2024-02-15:
Luckily we don't have to implement this given CODESYS 3.5 requires `()` on action calls, so I'm gonna go ahead and close it.

closed 2024-02-15 by volsa