PLC-lang/rusty #712: Entered unreachable code: left of an assignment must be a reference

volsa, 2022-12-27:
The following code 
```ST
FUNCTION func : DINT
   VAR_IN_OUT
       x : INT;
   END_VAR
END_FUNCTION

PROGRAM main
   func(1 := 1);
END_PROGRAM
```
will trigger the `unreachable!` line in https://github.com/PLC-lang/rusty/blob/a51d5b4095d4491907c92f5a6aeb5e4a27ecd2bc/src/codegen/generators/expression_generator.rs#L2443-L2466

99NIMI, 2022-12-28:
I edited your ST example, FUNCTIONS always need a return type

99NIMI, 2022-12-28:
The error is ok
You can only assign values to references/qualified-references

With the planned error handling refactor i would move this to the validation so we don't panic in codegen
if we find errors like this we probably shouldn't trigger the codegen
we can't generate anything valid for '1 := 1'

closed 2023-02-14 by mhasel