PLC-lang/rusty #719: Clearer error message for `Ambiguous datatype` errors

volsa, 2022-12-29:
The following code 
```ST
TYPE D : STRUCT 
    // ... 
END_STRUCT 
END_TYPE
```
produces the error message `D: Ambiguous datatype.` because D is an actual datatype, see https://github.com/PLC-lang/rusty/blob/25d7a9edf6f591a69dd8e3381053b4c33585da51/src/typesystem.rs#L61

I feel like the error message could be a bit clearer, e.g. something along 'D is a reserved keyword, can not use it as a name'

riederm, 2022-12-29:
Actually the validator tries to tell you, that there are two data types with the same name (so talking about bad messages :smile:). So it has nothing to do with a keyword.

Feel free to improve the message so it gets clearer. It uses the same text for duplicate datatypes, global variables, and POUs I think.

volsa, 2023-01-09:
I might have opened this issue prematurely, because I thought all error messages of this type would be like the error message for D as can be seen in the following image 😅 
[image]

I guess this could be improved for that edge-case but honestly the message are as clear as it gets, so I'd be fine with closing this issue (also my bad lol :P)

riederm, 2023-01-09:
wir sollten hier die internal-locations in den `addtional_locations` herausfiltern.
https://github.com/PLC-lang/rusty/blob/2e8d2ff5add11a9775bf97ff5bfa4105d872095d/src/diagnostics.rs#L795

das Problem passiert hier:

https://github.com/PLC-lang/rusty/blob/2e8d2ff5add11a9775bf97ff5bfa4105d872095d/src/diagnostics.rs#L805-L809

riederm, 2023-01-09:
if main_location is internal -> report with eprintln!(...) like we do now, just with a better message: `"An Error occured in an internal compiler location: <message>"`.
if an additional_location is internal -> skip this see-also

volsa, 2023-02-02:
Coming back to this because I just encountered it again, thinking maybe we could introduce a validator for function, struct, type,.. names checking against the internally reserved data type names? I'm assuming this should be relatively straightforward by extending the global validator? 
[image]

riederm, 2023-02-14:
was this covered by #748 ?

volsa, 2023-02-14:
Nope

PR #816 (2023-03-27): Report name conflicts
Find and report user-defined datatype names which clash with built-in datatypes, e.g. [image]

Fixes #719